jgromes/RadioLib #201: Stream implementation

jgromes, 2020-11-12:
As per discussion in #163 with @phretor and @rnbokade, it could be possible to extend maximum packet length beyond the FIFO size limitation by providing user with methods to feed the FIFO "on-the-fly", essentially creating a data stream instead of packetized transfer. This could be implemented for any module with interrupts on FIFO level, currently:

* CC1101
* Si443x/RFM2x
* SX127x/RFM9x in FSK mode

Transmitter (method names are placeholders):

1. In Arduino sketch, user creates an ISR, e.g.

```
void fifoAdd(void) {
  radio.fifoAdd(txBuffPtr, &remLength);
}
```

2. The ISR is passed as callback to e.g. `radio.setFifoEmptyAction(fifoAdd);`, much like `setFlag`/`setDio0Action`.
3. If the packet is longer than FIFO size, `startTransmit()` enables interrupt for some FIFO level. If not, interrupt is not enabled.
4. `startTransmit` begins transmission just like the current implementation and returns immediately after setting mode to Tx.
5. `fifoAdd` ISR now gets called every time FIFO level gets low enough, progressively filling FIFO with new bytes to be sent.
6. When `remLength` gets to zero, FIFO level interrupt is disabled and transmission is considered finished.

Receiver structure is the same, the only difference is that interrupt must be triggered on a different FIFO fill level (close to full), and the method called from ISR is extracting bytes from (e.g. `fifoGet(uint8_t* buffPtr, size_t* remLength)`).

rnbokade, 2020-11-13:
Thanks for looking into this. I will also be working side by side on implementation. BTW, I think RF69 will also be compatible with this approach, please confirm.

jgromes, 2022-07-10:
Finally got around to this, Stream is now implemented for SX127x. Have to say, watching the waterfall is quite funny when sending 512 byte packets using a 4800 bps transmission.

[image]

Will add support for the other modules mentioned above in the coming days.

closed 2022-07-10 by jgromes