jgromes/RadioLib #202: SX126x not working on custom PCB built upon SAMD51 adafruit M4 express

TinManAkshay, 2020-11-15:
Hi,

Our custom PCB is built on the MCU used on the Adafruit M4 express and we have SX1262 module on our board. We are also using the same bootloader as being used in adafruit feather. I am trying to run the example codes TX and RX on our custom board. But I am getting error failed code -2. So when you tried on Adafruit M4 express, did you make any changes on the open source variant.cpp and variant.h files to get the SX1262 working? Its because I did and had to introduce few new pins to the variant files on our new custom board. 

So what could be the issue. Any suggestions?

Thanks,
Akshay

jgromes, 2020-11-15:
I suggest to examine the documentation first.  Soon, you'll find that error code -2 is `ERR_CHIP_NOT_FOUND`, strongly pointing towards a hardware issue.

Next, you will find that there's a [handy wiki page](https://github.com/jgromes/RadioLib/wiki/Debug-mode) about Debug mode, which will allow you to examine the SPI transactions in detail, giving you more information about what exactly is happening instead of taking guesses about variant files. Post the debug output, the schematic and double-check your hardware configuration (pin setup, SPI setup etc.).

> So when you tried on Adafruit M4 express, did you make any changes on the open source variant.cpp and variant.h files to get the SX1262 working?

The readme file states that "Compilation of all examples is tested for all platforms prior to releasing new version." - I haven't tested every single module on every single platform, that would be hundreds of combinations.

TinManAkshay, 2020-11-15:
@jgromes , Now i am getting failed code -705. How will I know that this error is for what issue? Is it mentioned in any of the files?

jgromes, 2020-11-15:
It's on the [GitHub pages](https://jgromes.github.io/RadioLib/group__status__codes.html), linked from the readme. -705 is ERR_SPI_CMD_TIMEOUT; again, most likely a hardware issue.

TinManAkshay, 2020-11-15:
Doesnt it mean that we can do some changes in the firmware so that SPI timeout can be resolved? Earlier i was getting Chip not found error. That is gone

jgromes, 2020-11-15:
`ERR_CHIP_NOT_FOUND` and `ERR_SPI_CMD_TIMEOUT` are very similar, you can cause both of them by just running the code on an MCU with no SX126x connected. `ERR_SPI_CMD_TIMEOUT` can be triggered when the BUSY line state doesn't change, `ERR_CHIP_NOT_FOUND` can be caused when all the SPI reads is 0xFF - a classic example of disconnected SPI pins.

> Doesnt it mean that we can do some changes in the firmware so that SPI timeout can be resolved?

Maybe, I don't have the hardware and from the information you have posted here (or rather lack thereof), there's no way for me to guess. I suggest going back to my first comment and **posting what I asked for - the debug mode output, the schematic, and the code.** 

I also can't guarantee RadioLib will work on your custom platform, since you mentioned you made some changes to the variant files, so maybe it would be easier to get it running on a setup with known working hardware before attempting to design your own.

TinManAkshay, 2020-11-15:
Code is this...

/*
   RadioLib SX126x Transmit Example

   This example transmits packets using SX1262 LoRa radio module.
   Each packet contains up to 256 bytes of data, in the form of:
    - Arduino String
    - null-terminated char array (C-string)
    - arbitrary binary data (byte array)

   Other modules from SX126x family can also be used.

   For default module settings, see the wiki page
   https://github.com/jgromes/RadioLib/wiki/Default-configuration#sx126x---lora-modem

   For full API reference, see the GitHub Pages
   https://jgromes.github.io/RadioLib/
*/

// include the library
#include <RadioLib.h>

// SX1262 has the following connections:
// NSS pin:   4
// DIO1 pin:  13
// NRST pin:  11
// BUSY pin:  12

  SX1262 radio = new Module(5, 13, 11, 12);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1262 radio = RadioShield.ModuleA;

void setup() {
  Serial.begin(9600);
  pinMode(PIN_LORA_PWREN, OUTPUT);
  pinMode(11, OUTPUT);
  pinMode(12, OUTPUT);
    
  digitalWrite(PIN_LORA_PWREN, HIGH); 
  digitalWrite(11, LOW); //reset
  digitalWrite(12, LOW); //busy

   delay(500);

  digitalWrite(11, HIGH); //reset 
  
  // initialize SX1262 with default settings
  Serial.print("[SX1262] Initializing ... ");
  int state = radio.begin(915.0, 500.0, 7, 4, 0x34, 20);
 
  if (state == ERR_NONE) {
    Serial.println("success!");
  } else {
    Serial.print("failed, code ");
    Serial.println(state);
    while (true);
  }

  // some modules have an external RF switch
  // controlled via two pins (RX enable, TX enable)
  // to enable automatic control of the switch,
  // call the following method
  // RX enable:   4
  // TX enable:   5
  /*
    radio.setRfSwitchPins(4, 5);
  */
}

void loop() {
  Serial.print(F("[SX1262] Transmitting packet ... "));

  // you can transmit C-string or Arduino string up to
  // 256 characters long
  // NOTE: transmit() is a blocking method!
  //       See example SX126x_Transmit_Interrupt for details
  //       on non-blocking transmission method.
 // int state = radio.transmit("Hello World!");

  // you can also transmit byte array up to 256 bytes long
  
    byte byteArr[] = {0x01, 0x23, 0x45, 0x56, 0x78, 0xAB, 0xCD, 0xEF};
    int state = radio.transmit(byteArr, 8);
  

  if (state == ERR_NONE) {
    // the packet was successfully transmitted
    Serial.println(F("success!"));

    // print measured data rate
    Serial.print(F("[SX1262] Datarate:\t"));
    Serial.print(radio.getDataRate());
    Serial.println(F(" bps"));

  } else if (state == ERR_PACKET_TOO_LONG) {
    // the supplied packet was longer than 256 bytes
    Serial.println(F("too long!"));

  } else if (state == ERR_TX_TIMEOUT) {
    // timeout occured while transmitting packet
    Serial.println(F("timeout!"));

  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }

  // wait for a second before transmitting again
  delay(500);
}

This is what I am getting on the Serial monitor. 

[image]

TinManAkshay, 2020-11-15:
The schematic is here ..

[image]

[image]

[image]

jgromes, 2020-11-15:
Now we're getting somewhere ;)

1. In your sketch, you have taken manual control of some pins, namely RST and BUSY - don't do that, the library will handle them. That's not the main issue here, but I wanted to point that out.

2. The reset input of SX126x devices is active low, you have a pulldown on it (R52). That resistor will keep the device in reset forever, hence no communication. Remove R52 and check again.

Other than that, I don't see anything immediately obvious, though I don't know the SAMD platform well enough to for example check the SPI pins and SPI setup in the Arduino sketch.

TinManAkshay, 2020-11-15:
I have removed the manual pins as you pointed out.

PCB is with me actually. I can only make changes to the firmware or any other file. Unfortunately, I cannot make any hardware changes. I have tried to keep RESET pin HIGH, but same error shows up.

jgromes, 2020-11-15:
Sorry, but this really seems like a hardware issue with your design. From the debug output it looks like the module "responds" with 0x00 to the very first command (though that can also be the result of disconnected pins), and then the program times out waiting for BUSY line to go through a high to low transition. From that my guess is that either:

1. BUSY line is disconnected or connected to the wrong pin or
2. The module isn't powered.

Either way you're going to need to find out what exactly is going on with the BUSY line (as well as checking that all the other pins such as SPI have the expected signals), so I'm closing this for now as it doesn't seem relevant to the library code at this point.

closed 2020-11-15 by jgromes