jgromes/RadioLib #203: SX126x radio = new module()

TinManAkshay, 2020-11-19:
Hi,

In the example, I replaced SX1262 radio = new module() with SX126x radio=new module(). But I can the setRffrequency() is protected inside the class inside SX126x.h file. So how i can set the frequency?

FYI: I used SX126x radio because SX1262 radio was not giving success while initializing the radio settings. So i got success with Sx126x radio.

-Akshay

jgromes, 2020-11-19:
The SX126x class isn't meant to be used from the Arduino sketch, you have to use one of the derived classes like SX1268 or SX1262.  SX126x isn't giving you errors probably because the incorrect parameter isn't being configured at all.

It's not a good idea to try side-step security checks within the library - if it's giving an error, there's a reason for it. What's the error message it was giving you?

TinManAkshay, 2020-11-19:
I was getting failed code -707. I saw by uncommenting out the VERBOSE and DEBUG that all the traffic flow between mcu and lora chip. It was correct. So while calling...

  int state = radio.begin(923.3, 500.0, 9, 7, 0x34, 13, 8, 3.3, false)

I am getting the error code -707.

jgromes, 2020-11-19:
Error code -707 is returned by the SX126x chip itself when there's something wrong with the command or its execution - unfortunately SX126x docs aren't very clear on the exact cause.

There's a known issue with SPI communication to SX126x on some faster platforms (#158). There's a patch for it, but your platform must define `SAMD_SERIES` macro.

TinManAkshay, 2020-11-19:
I saw some samd declaration inside the buildopt.h and where i should define the SAMD macro? Inside the arduino sketch?

I have attached file where all the verbose lines are there.

[707 error.txt](https://github.com/jgromes/RadioLib/files/5569762/707.error.txt)

jgromes, 2020-11-19:
`SAMD_SERIES` is a macro that should be defined by the platform - I don't recall where exactly is it defined, could be in boards.txt, platform.txt or even by the compiler.

You can check which platform RadioLib thinks it's running on by enabling debug mode and verbose output during compilation in Arduino IDE -> Preferences. During compilation, it will print the platform it found to the console. This is output on my machine when compiling for Arduino Zero:

```
Compiling sketch...
"C:\\Users\\jangr\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD -DF_CPU=48000000L -DARDUINO=10813 -DARDUINO_SAMD_ZERO -DARDUINO_ARCH_SAMD -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804d -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Zero\"" "-IC:\\Users\\jangr\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-IC:\\Users\\jangr\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS-Atmel\\1.2.0/CMSIS/Device/ATMEL/" "-IC:\\Users\\jangr\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.9\\cores\\arduino" "-IC:\\Users\\jangr\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.9\\variants\\arduino_zero" "-IC:\\Users\\jangr\\Documents\\Arduino\\libraries\\RadioLib\\src" "-IC:\\Users\\jangr\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.9\\libraries\\SPI" "C:\\Users\\jangr\\AppData\\Local\\Temp\\arduino_build_909562\\sketch\\SX127x_Transmit.ino.cpp" -o "C:\\Users\\jangr\\AppData\\Local\\Temp\\arduino_build_909562\\sketch\\SX127x_Transmit.ino.cpp.o"
In file included from C:\Users\jangr\Documents\Arduino\libraries\RadioLib\src/TypeDef.h:4:0,
                 from C:\Users\jangr\Documents\Arduino\libraries\RadioLib\src/RadioLib.h:43,
                 from C:\Users\jangr\Documents\Arduino\libraries\RadioLib\examples\SX127x\SX127x_Transmit\SX127x_Transmit.ino:20:
C:\Users\jangr\Documents\Arduino\libraries\RadioLib\src/BuildOpt.h:153:57: note: #pragma message: RADIOLIB_PLATFORM: Arduino/Adafruit SAMD
     #define RADIOLIB_PLATFORM                           "Arduino/Adafruit SAMD"
                                                         ^
C:\Users\jangr\Documents\Arduino\libraries\RadioLib\src/RadioLib.h:55:41: note: in expansion of macro 'RADIOLIB_PLATFORM'
   #pragma message "RADIOLIB_PLATFORM: " RADIOLIB_PLATFORM
                                         ^~~~~~~~~~~~~~~~~
Compiling libraries...
Compiling library "RadioLib"
```

Checking the log, it's failing at command 0x86 (SetRfFrequency) or 0x98 (CalibrateImage) - that's why it didn't fail when you tried directly creating an instance of SX126x, since it doesn't configure frequency.

923.3 MHz should be within valid range for SX1262, but could you try a different frequency, like 434 MHz?

TinManAkshay, 2020-11-19:
I am in US, so we have to use US band right? 902-927mhz

TinManAkshay, 2020-11-19:
Also, I tried writing a simple spi write test code to set the frequency and calibrate it. To every write, I got 0xA2 result out that means write is successful right. Somehow its not liking the frequency parameter. I strongly believe SPI signals are operating correctly otherwise commands being set earlier than setFrequency would not have been successful.

-Akshay

abarangan, 2020-11-20:
@jgromes , over here helping @TinManAkshay with this problem. The code was initializing with `radio.begin(915.0, 250.0, 7, 5, 0x34, 20);` but I tried the default `radio.begin();` in case it's a configuration issue (and to attempt 434MHz) and found the same "Failure to execute" error returned by the part at the same location.

The `Module::delay(1);` workaround for #158 is being executed so that doesn't appear to be the issue.

I'll keep digging.

jgromes, 2020-11-20:
@abarangan OK, thanks - next thing that comes to mind with frequency and calibration issues is the crystal - are you using TCXO or plain XTAL?

abarangan, 2020-11-20:
@jgromes, Can't thank you enough. I hadn't noticed the TCXO/XTAL setting that was in the code.  I've changed initialization to:

`radio.begin(915.0, 250.0, 7, 5, 0x34, 20, 10, 0, false);`

and things are working fine.

Thanks for your support. I'm impressed with the structure of your library.

closed 2020-11-20 by jgromes