jgromes/RadioLib #204: Remove reset in findChip

kiryam, 2020-11-20:
I little bit confused about reset chip in findChip method 
https://github.com/jgromes/RadioLib/blob/master/src/modules/SX127x/SX127x.cpp#L1116

I think reset method should be called explicit.

jgromes, 2020-11-20:
I disagree - it's common practice to reset a peripheral during initialization, because it will put it into a known state. It can also help resolve issues with locked-up communication bus, which may not be possible to solve otherwise (though that's not common on SPI).

Wha do you think reset should be called explicitly?

kiryam, 2020-11-20:
I expected from method findChip version of module. May be we can split this method to findChip and getChipVersion

jgromes, 2020-11-20:
Not really, it's supposed to find the chip itself - i.e. verify that it is connected to the MCU and that it really is the chip the user says it should be, which is what the version register check is for. It's not supposed to provide the version number to code outside the library.

Do I understand correctly that you want to access the version number outside the library code? If so, why? And if not, what exactly are you trying to do?

kiryam, 2020-11-20:
Yes, it is. I use library in "godmode". Also I use deepsleep mode. And after wakeup I tries to verify is module still works. (device works from battery and module may not have enough voltage)

jgromes, 2020-11-20:
OK, in that case it's going to be better to expose some sort of `getChipVersion` - probably also a good idea to have it available for more than just SX127x, a lot of the radios have some version register.

jgromes, 2020-11-20:
`getChipVersion()` method to read version register was added for all modules that support this (CC1101, RF69, Si443x/RFM2x, SX127x/RFM9x).

@kiryam on a side note, you really shouldn't be using godmode as a workaround - if there's a feature you feel is missing, open an issue instead.

closed 2020-11-20 by jgromes