jgromes/RadioLib #205: SNR on SX1276 is not accurate.

mc-hamster, 2020-11-22:
Calling SX127x::getSNR() most often returns a value between 5 and 6. Every once in a while it'll return something outside that range, but very very rarely.

SX126x::getSNR() operates as it should.

As an example, when receiving a packet between two lora devices, I got:

RX SNR 5.000000
RX RSSI -59.000000

With a 50db signal attenuator installed in series with a TX antenna, the RX device got:

RX SNR 5.500000
RX RSSI -94.000000

The noise floor of my living room at the moment was about -87db.

This bug was tracked down while researching:

https://github.com/meshtastic/Meshtastic-device/issues/332

jgromes, 2020-11-22:
I think the SNR value should be accurate, for two reasons:

1. There's not really that much going on when reading SNR, it's literally just one SPI register read, then divide by 4 - I don't see how that can be inaccurate.

2. I don't know why SNR is so large in your setup, but I know that negative values are common. As an example, see https://fossa.apaluba.com/nodered/dashboard/packet/33624938e1c7b9d8c08d4cc24dd24f57cd0b52ed - that's a LoRa packet that was transmitted from the Norbi satellite, and it shows that most ground stations report negative SNR.

Please post your setup, hardware- and software-wise, as per the issue templates.

mc-hamster, 2020-11-22:
I was expecting to see more negative values. I'll post sample code, hardware, etc shortly. 

Sorry for not following the template. Busy fingers :)

Cheers!

mc-hamster, 2020-11-22:
On second thought, I’m going to close this for now. I’m getting a jtag debugger that’ll help me pin point where this is coming from. 

Sorry for opening the bug and thanks for the quick response. Cheers!

closed 2020-11-22 by mc-hamster