jgromes/RadioLib #206: Receive command from SX1272 to SX1262

TinManAkshay, 2020-11-23:
Hi,

So far we have been successful with the data transmission from SX1262 to SX1272 but when I try to send the data packet from SX1272 to SX1262, SX1262 doesnt receive. I have seen the dual communication between two SX1262 boards and two SX1272 boards, keeping the configuration is same. But SX1262 is not receiving any command from SX1272. Does anybody have a clue about this?

-Akshay

TinManAkshay, 2020-11-23:
Also, I am using Receiver code (blocking method) from the examples to receive on SX1262 side.

jgromes, 2020-11-23:
Once again you have posted very little information about your setup, making it nearly impossible for anybody to have a clue about this. Here's a few starter bits that should have been a part of the very first comment:

What are the exact modules you're using? Do they have appropriate antennas and power source? Can you independently verify that the SX127x is really transmitting (e.g. with an SDR or another SX127x)?

>  I have seen the dual communication between two SX1262 boards and two SX1272 boards, keeping the configuration is same.

What is the configuration? Are you using LoRa or FSK mode?

> I am using Receiver code (blocking method) from the examples to receive on SX1262 side.

What is running on the SX127x side?

TinManAkshay, 2020-11-23:
I am sorry for not being more specific. 

1. As you already know that I have a custom board (SX1262 module on it) and Dragino board (SX1272 on it). Earlier we used to have dual communication between Adafruit Atmega 32u4 (SX1272) and Dragino. Now we want to have dual comm between SX1262 and SX1272. 

2. Yes I am 100% sure that SX1272 is transmitting the packet in the form of bytes as it worked before between two SX1272 modules. 

3. Yes they have antennas. I have connected SX1262 board to the 5V USB and SX1272 dragino is also getting proper power supply. 

4. On both sides, I have kept, 923.3 Mhz, SF=7, Coding rate = 4/5, Preamble length=8, Power=20. This is LoRa actually.

5. I am using Adafruit_rfm9x library on SX1272 dragino board.

Thanks,
Akshay

jgromes, 2020-11-23:
OK, have you tried running RadioLib code on both ends? I don't know what sort of configuration the Adafruit library does. There's more LoRa parameters than just those you have listed, such as sync word or header type,

TinManAkshay, 2020-11-23:
No I haven't tried radiolib on SX1272 as Dragino is a LoRa HAT for Raspberry Pi. On the Raspberry Pi, Adafruit library is built on Python. 
Bandwidth is 500khz on both ends. No sync mode is set for LoRa in Adafruit's library. Since SX1272 is receiving the packets from SX1262, i dont think thats an issue. What do you say about that?
Also, how should I check the tx_header in SX1262 library? Does it make a difference in two different LoRa modules?

-Akshay

jgromes, 2020-11-23:
You haven't mentioned the SX127x is running on Raspberry (there are also Dragino shields for Arduino, I assumed you had one of those).

> No sync mode is set for LoRa in Adafruit's library.

That doesn't seem right - LoRa packets will always have some sync word, either 0x12 for private networks, or 0x34 for LoRaWAN. Sync word in LoRa behaves a lot differently than it does in FSK, it's not a 100% guarantee of packet filtering, but it still should be set to the same value.

SX126x driver in RadioLib is using explicit header mode by default - you can also check the other parameters that are configured on startup on [this wiki page](https://github.com/jgromes/RadioLib/wiki/Default-configuration#sx126x---lora-modem).

Unforuntunately I can't really guarantee RadioLib will work with other SX127x libraries, though I'm fairly confident different setups running RadioLib can send LoRa packets from SX1272 and receive them on SX1262 and vice-versa. So you'll have to dig into the Adafruit Python library and find out what configurations is it doing when it's transmitting packets.

Also, have you considered the following statement in the Adafruit library?

```
    Also note this library tries to be compatible with raw RadioHead Arduino
    library communication. This means the library sets up the radio modulation
    to match RadioHead's defaults and assumes that each packet contains a
    4 byte header compatible with RadioHead's implementation.
```

from https://github.com/adafruit/Adafruit_CircuitPython_RFM9x/blob/6590c91a9146f41a7984a52febce9b21d0df8748/adafruit_rfm9x.py#L154

closed 2020-11-23 by jgromes