jgromes/RadioLib #212: CC1101 add receive timeout

13, 2020-12-09:
Could you add a receive timeout?

I had issues after ~1-2 hours, but with this quick & dirty fix I was able to resolve it.

```
int16_t CC1101::receive(uint8_t* data, size_t len) {
  // add timeout 5 seconds
  uint32_t timeout = 5e6;

  // start reception
  int16_t state = startReceive();
  RADIOLIB_ASSERT(state);

  // wait for sync word or timeout
  uint32_t start = micros();
  while(!Module::digitalRead(_mod->getIrq())) {
    Module::yield();
    if(micros() - start > timeout) {
      // flush Rx FIFO
      SPIsendCommand(CC1101_CMD_FLUSH_RX);
      // clear internal flag so getPacketLength can return the new packet length
      _packetLengthQueried = false;
      // set mode to standby
      standby();
      return(ERR_RX_TIMEOUT);
    }
  }

```

jgromes, 2020-12-09:
That's a good point - there should be timeout in the `while` loops that wait for pin state changes - similarly to how it's handled in the other modules.

Constant 5 second timeout is probably not the best way to do that though - CC1101 can achieve very low transmission speed, down to 25 bps. Instead I'll add timeouts based on expected (or maximum) packet length and currently configured data rate.

jgromes, 2021-02-12:
Added in 4461e9a98e4b33accf93c64c43cb6f8aa2c025d8

closed 2021-02-12 by jgromes