jgromes/RadioLib #213: Issue with RTTY on Adafruit Feather Lora Module SX1278

rifath-shaarook, 2020-12-16:
I've been trying to use RTTY and Morse code on the Adafruit Feather 32u4 board with SX1278 (https://www.adafruit.com/product/3079) which is a 433 MHz Version. The board is working fine with normal transmit code but when it comes to RTTY and Morse, it's sometimes transmitting, sometimes showing error -16, and sometimes not showing anything on the serial monitor, I'm pretty sure that the SPI wiring is perfect because it's on the PCB itself and normally transmit code works just fine. Like I said that it's sometimes working, it's only during the first boot after uploading the code, if it presses the reset button and remove the power and reconnect then it's not working, that too only if Serial Monitor is open. Not able to figure out what would be the exact issue. I'm just using the example code from the library. Any help would be really appreciated.

Tried with Various versions of Arduino including the latest one.

Using Adafruit Feather 32u4 board with on-board RFM96 which actually has SX1278 inside.

here's the code with modified SPI pins for this particular board.

/*
   RadioLib RTTY Transmit Example

   This example sends RTTY message using SX1278's
   FSK modem.

   Other modules that can be used for RTTY:
    - SX127x/RFM9x
    - RF69
    - SX1231
    - CC1101
    - SX126x
    - nRF24
    - Si443x/RFM2x
    - SX128x

   For default module settings, see the wiki page
   https://github.com/jgromes/RadioLib/wiki/Default-configuration

   For full API reference, see the GitHub Pages
   https://jgromes.github.io/RadioLib/
*/

// include the library
#include <RadioLib.h>
int state;
// SX1278 has the following connections:
// NSS pin:   10
// DIO0 pin:  2
// RESET pin: 9
// DIO1 pin:  3
SX1278 radio = new Module(8, 7, 4, 11);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1278 radio = RadioShield.ModuleA;

// create RTTY client instance using the FSK module
RTTYClient rtty(&radio);

void setup() {
  Serial.begin(9600);
  //delay(3000);
  // initialize SX1278 with default settings
  while(true){
  Serial.print(F("[SX1278] Initializing ... "));
  state = radio.beginFSK();

  // when using one of the non-LoRa modules for RTTY
  // (RF69, CC1101, Si4432 etc.), use the basic begin() method
  // int state = radio.begin();

  if(state == ERR_NONE) {
    Serial.println(F("success!"));
    break;
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    delay(1000);
    //while(true);
  }
  }

  

  // initialize RTTY client
  // NOTE: RTTY frequency shift will be rounded
  //       to the nearest multiple of frequency step size.
  //       The exact value depends on the module:
  //         SX127x/RFM9x - 61 Hz
  //         RF69 - 61 Hz
  //         CC1101 - 397 Hz
  //         SX126x - 1 Hz
  //         nRF24 - 1000000 Hz
  //         Si443x/RFM2x - 156 Hz
  //         SX128x - 198 Hz
  Serial.print(F("[RTTY] Initializing ... "));
  // low ("space") frequency:     434.0 MHz
  // frequency shift:             183 Hz
  // baud rate:                   45 baud
  // encoding:                    ASCII (7-bit)
  // stop bits:                   1
  state = rtty.begin(434.0, 183, 45);
  if(state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while(true);
  }

  /*
    // RadioLib also provides ITA2 ("Baudot") support
    rtty.begin(434, 183, 45, ITA2);

    // All transmissions in loop() (strings and numbers)
    // will now be encoded using ITA2 code

    // ASCII characters that do not have ITA2 equivalent
    // will be sent as NUL (including lower case letters!)
  */
}

void loop() {
  Serial.print(F("[RTTY] Sending RTTY data ... "));

  // send out idle condition for 500 ms
  rtty.idle();
  delay(500);

  // RTTYClient supports all methods of the Serial class

  // Arduino String class
  String aStr = "Arduino String";
  rtty.println(aStr);

  // character array (C-String)
  rtty.println("C-String");

  // string saved in flash
  rtty.println(F("Flash String"));

  // character
  rtty.println('c');

  // byte
  // formatting DEC/HEX/OCT/BIN is supported for
  // any integer type (byte/int/long)
  rtty.println(255, HEX);

  // integer number
  int i = 1000;
  rtty.println(i);

  // floating point number
  float f = -3.1415;
  rtty.println(f, 3);

  Serial.println(F("done!"));

  // wait for a second before transmitting again
  delay(1000);
}

jgromes, 2020-12-16:
Regarding your Serial connection issue, I think that's because you're using ATmega32u4, which has native USB (unlike the more "classic" ATmega328P), so you might run into issues if you're doing `Serial.print`s while USB is disconnected.

Next:
> with on-board RFM96 which actually has SX1278 inside.

That claim is not entirely true - RFM9x have some differences over SX127x. Please use the correct radio type in your code.

Finally, please post the debug output from your sketch that's producing the -16 error codes. The one above seems to only run the `beginFSK()` method in an infinite loop. See [this wiki page](https://github.com/jgromes/RadioLib/wiki/Debug-mode) for details on debug mode, enable both debug levels and post the Serial output.

rifath-shaarook, 2020-12-18:
Thank you so much for the quick response. I have enabled the Debug and Verbose and found some interesting outcomes.

When I uncomment both or just the Verbose, the module is just working like a charm, whether the serial monitor is open or not or if it is connected to the computer or a battery, the radio is working and I have checked it with SDR.

When I uncomment just the Debug and commented on the Verbose, the board is working but only when I open the serial monitor.

When I uncomment both the Debug and Verbose, I'm getting Error Code -16.

Found it really interesting, would be great if you help us find why it is happening and what would be the right solution, we'll also try to do the same. Checked all the scenarios with SDR. Added the verbose output in the attachment, not able to get the debug output since it is not showing even when I just uncommented the debug, it's slightly showing transmitting RTTY while opening the serial port.

[debug_output.txt](https://github.com/jgromes/RadioLib/files/5715181/debug_output.txt)

jgromes, 2020-12-19:
I can't really reproduce this on my end - I don't have any ATmega32u4 boards on hand.

Could you double-check the pinout? Your sketch is using pin 11 for DIO1, I can't find that information on https://learn.adafruit.com/assets/46253 You can use `RADIOLIB_NC` as alias for unused pin.

Also, it might be worth checking what's the influence of the reset pin - use `RADIOLIB_NC` isntead of the reset pin number.

jgromes, 2021-01-15:
Closed due to inactivity.

closed 2021-01-15 by jgromes

rifath-shaarook, 2021-01-19:
My Issue has been solved with enabling the verbose, I guess the problem is to do with Serial function of 32u4 mcu. Will let you know if I'm able to find the right reason.