jgromes/RadioLib #218: Can't compile for Arduino ATMEGA4809

ok1cdj, 2020-12-25:
**ERROR when compiling**
`In file included from /home/ondra/Arduino/libraries/RadioLib/src/modules/HC05/HC05.h:4:0,
                 from /home/ondra/Arduino/libraries/RadioLib/src/RadioLib.h:65,
                 from /home/ondra/Arduino/rfm69-rtty/rfm69-rtty.ino:17:
/home/ondra/Arduino/libraries/RadioLib/src/modules/HC05/../../ISerial.h:30:12: error: 'size_t ISerial::print(const char*)' cannot be overloaded
     size_t print(const char[]);
            ^~~~~
/home/ondra/Arduino/libraries/RadioLib/src/modules/HC05/../../ISerial.h:28:12: error: with 'size_t ISerial::print(const char*)'
     size_t print(const __FlashStringHelper *);
            ^~~~~
/home/ondra/Arduino/libraries/RadioLib/src/modules/HC05/../../ISerial.h:42:12: error: 'size_t ISerial::println(const char*)' cannot be overloaded
     size_t println(const char[]);
            ^~~~~~~
/home/ondra/Arduino/libraries/RadioLib/src/modules/HC05/../../ISerial.h:40:12: error: with 'size_t ISerial::println(const char*)'
     size_t println(const __FlashStringHelper *);
            ^~~~~~~`

**To Reproduce**
Select board ATMEGA4809 and try use library and compile

**Additional info:**
 - MCU: ATMEGA4809
 - Link to Arduino core: https://github.com/MCUdude/MegaCoreX
 - Wireless module type: RFM96W
 - Arduino IDE version  1.8.5
 - Library version 4.0.6

jgromes, 2020-12-25:
MegaCoreX isn't supported at the moment, seems like there's some slight incompatibility between that and standard Arduino cores.

jgromes, 2020-12-25:
Support added in 65379c477233732a08d856462155deb64aa3ee9d - no idea if it works, but all examples do compile.

closed 2020-12-25 by jgromes

ok1cdj, 2020-12-25:
How to set SPI pins, Atmega4809 have 3 options for SPI:

SPI	swap(0) or pins(4,5,6,7)	swap(1) or pins(14,15,16,17)	swap(2) or pins(30,31,32,33)

SPIClass works different here.

jgromes, 2020-12-25:
Admittedly I have no experience with MegaCoreX, but I think you should be able to just call SPI.pins before any calls to RadioLib methods such as radio.begin(), right?

ok1cdj, 2020-12-26:
@jgromes thank you, confirm that library works with MegaCoreX. Tested with RFM69 and ATMega4809.