jgromes/RadioLib #230: Second Module not working on the Portenta board

semir-tursunovic, 2021-01-13:
`int state = radio_tx.begin(RADIO_TX_FREQ, 125.0, 9, 7, SX126X_SYNC_WORD_PRIVATE, 10, 8, 1.6, false);
	g_RADIO_TX.frequency = RADIO_TX_FREQ;
	if (state == ERR_NONE)
	{
		printUART("success!\n");
	}
	else
	{
		printUART("failed, code %d\n",state);
		//while (true);
	}
	
	radio_tx.setDio1Action(irqTxRADIO);
	printUART("-> RADIO: tx ready...\n");
	//state = radio_rx.startReceive();
	//if (state == ERR_NONE)
	//{
		//printUART("success!\n");
	//}
	//else
	//{
		//printUART("failed, code %d\n",state);
		////while (true);
	//}
	delay(20);
	g_RADIO_RX.state = 0x00;
	printUART("-> RADIO: init rx... ");
	state = radio_rx.begin(434.0, 125.0, 9, 7, SX126X_SYNC_WORD_PRIVATE, 10, 8, 1.6, false);
	g_RADIO_RX.frequency = RADIO_RX_FREQ;
	if (state == ERR_NONE)
	{
		printUART("success!\n");
	}
	else
	{
		printUART("failed, code %d\n",state);
		//while (true);
	}
	radio_rx.setDio1Action(irqRxRADIO);
	printUART("-> RADIO: rx listening... ");
	state = radio_rx.startReceive();
	if (state == ERR_NONE)
	{
		printUART("success!\n");
	}
	else
	{
		printUART("failed, code %d\n",state);
		//while (true);
	}`

**Hardware setup**
We are using two SC126x based modules. One for Tx and second one for Rx. If we initialize modules separately they work fine. But when we try the above code for the second one we get error status -707. Both module are sharing the SPI peripheral with separately controlled CS line. 

**Additional info (please complete):**
 - MCU: Arduino Portenta H7
 - Link to Arduino core:https://github.com/arduino/ArduinoCore-mbed/
 - Wireless module type [SX1268]
 - Arduino IDE version  [e.g. 1.8.13]
 - Library version [e.g. 4.1.0]

jgromes, 2021-01-13:
Please post the entire sketch - especially the part where you're creating the module instances.

-707 is `ERR_SPI_CMD_FAILED`, which can be either returned by the module if the SPI communication is working correctly, or returned by the library if the SPI is only reading 0x00 or 0xFF bytes. From what you've described, I'd say the second option is more likely, but it's impossible to tell without the whole sketch.

You might also want to check the debug output.

semir-tursunovic, 2021-01-14:
The problem was in the SPI driver not the library. After driver was fixed it is working now. Thanks.

closed 2021-01-14 by jgromes