jgromes/RadioLib #242: Not receiving if SF changed at runtime

HamzaHajeir, 2021-02-03:
**Describe the bug**
Hi

I'm facing an issue of receiving messages when I change SF to the target SF I expect to receive messages by.

The change is using radio.setSpreadingFactor();

**To Reproduce**

Sender code (Always send messaged in SF 7):

```cpp
// include the library
#include <RadioLib.h>
#define D1  5
#define D0 16
#define D2 4

const int csPin = D0;    // LoRa radio chip select
const int resetPin = D2; // LoRa radio reset
const int irqPin = D1;   // change for your board; must be a hardware interrupt pin
const int DIO1 = 10;     // change for your board; must be a hardware interrupt pin
const int DIO0 = irqPin; // change for your board; must be a hardware interrupt pin

// SX1278 has the following connections:
// NSS pin:   5 D1
// DIO0 pin:  16 D0
// RESET pin: 4 :D2
// DIO1 pin:  10 SD3
SX1278 radio = new Module(csPin, DIO0, resetPin, DIO1);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1278 radio = RadioShield.ModuleA;

// save transmission state between loops
int transmissionState = ERR_NONE;

void setup() {
  Serial.begin(115200);

  // initialize SX1278 with default settings
  Serial.print(F("[SX1278] Initializing ... "));
  int state = radio.begin(434.0, 125.0, 7);
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  // set the function that will be called
  // when packet transmission is finished
  radio.setDio0Action(setFlag);

  // start transmitting the first packet
  Serial.print(F("[SX1278] Sending first packet ... "));

  // you can transmit C-string or Arduino string up to
  // 256 characters long
  transmissionState = radio.startTransmit("Hello World!");

  // you can also transmit byte array up to 256 bytes long
  /*
    byte byteArr[] = {0x01, 0x23, 0x45, 0x67,
                      0x89, 0xAB, 0xCD, 0xEF};
    state = radio.startTransmit(byteArr, 8);
  */
}
[... 405 lines cut ...]
> 13:53:15.488 > [SX1278] Frequency error:        -1166.02 Hz
> 13:53:16.519 > ReceivedFlag, Reading
> 13:53:16.526 > [SX1278] Received packet!
> 13:53:16.526 > [SX1278] Data:           ����������
> 13:53:16.526 > [SX1278] RSSI:           -63.00 dBm
> 13:53:16.529 > [SX1278] SNR:            9.50 dB
> 13:53:16.531 > [SX1278] Frequency error:        -1166.02 Hz
> 13:53:17.402 > Changing SF to 9
> 13:53:27.406 > Changing SF to 7
> 13:53:37.407 > Changing SF to 9
> 13:53:47.408 > Changing SF to 7
> 13:53:57.409 > Changing SF to 9
> 13:54:07.411 > Changing SF to 7
> 13:54:17.412 > Changing SF to 9
> 13:54:27.413 > Changing SF to 7
> 13:54:37.414 > Changing SF to 9
> 13:54:47.415 > Changing SF to 7
> 13:54:57.417 > Changing SF to 9
> 13:55:07.418 > Changing SF to 7

**Additional info (please complete):**
 - MCU:ESP8266: NodeMCU
 - Wireless module type: SX1278
 - Arduino IDE version  1.8.13
 - Library version 4.1.0

jgromes, 2021-02-03:
When you change modem configuration (such as spreading factor), the module is put into standby to not mess up any on-going transmission or reception. That will also stop the receive mode, so call `startReceive()` again after you change the spreading factor.

HamzaHajeir, 2021-02-03:
> When you change modem configuration (such as spreading factor), the module is put into standby to not mess up any on-going transmission or reception. That will also stop the receive mode, so call `startReceive()` again after you change the spreading factor.

Hi

Thank you for the quick response, I've missed this point.
I'll retest asap and inform with results.

Thanks

jgromes, 2021-02-18:
Closed due to inactivity, feel free to reopen later.

closed 2021-02-18 by jgromes

HamzaHajeir, 2021-02-18:
Missed to feedback. It works properly.