jgromes/RadioLib #244: MQTT Publish -- Error 'wifi' was not declared in this scope

nkebx2, 2021-02-08:
Hi. Thank you for sharing the library. I'm trying to use ESP8266 nodemcu with SX1278 radiolib for mqtt publishing and subscribing but I encounter this at the very beginning of the original codes.

```c++
// include the library
#include <RadioLib.h>

// ESP8266 has the following connections:
// TX pin: 9
// RX pin: 8
ESP8266 wifi = new Module(9, 8);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//ESP8266 wifi = RadioShield.ModuleA;

// create MQTT client instance using the wifi module
// the default port used for MQTT is 1883
MQTTClient mqtt(&wifi, 1883);

void setup() {
  Serial.begin(9600);

  // initialize ESP8266
  Serial.print(F("[ESP8266] Initializing ... "));
  // baudrate:  9600 baud
  int state = wifi.begin(9600);
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  // join access point
  Serial.print(F("[ESP8266] Joining AP ... "));
  // name:      SSID
  // password:  password
  state = wifi.join("SSID", "password");
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  // connect to MQTT server
  Serial.print(F("[ESP8266] Connecting to MQTT server ... "));
  // server URL:  broker.shiftr.io
  // client ID:   arduino
  // username:    try
  // password:    try
  state = mqtt.connect("broker.shiftr.io", "arduino", "try", "try");
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
}

void loop() {
  // publish MQTT message
  Serial.print(F("[ESP8266] Publishing MQTT message ... "));
  // topic name:            hello
  // application message:   world
  int state = mqtt.publish("hello", "world");
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
  }

  // wait for a second before publishing again
  delay(1000);
}
```
``

Error messages
`
Arduino: 1.8.13 (Windows 7), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

<command-line>:0:9: error: expected unqualified-id before numeric constant

C:\Users\User\Desktop\MQTT_Publish\MQTT_Publish.ino:22:1: note: in expansion of macro 'ESP8266'

 ESP8266 wifi = new Module(1, 3);

 ^

MQTT_Publish:30:18: error: 'wifi' was not declared in this scope

 MQTTClient mqtt(&wifi, 1883);

                  ^

C:\Users\User\Desktop\MQTT_Publish\MQTT_Publish.ino: In function 'void setup()':

MQTT_Publish:38:15: error: 'wifi' was not declared in this scope

   int state = wifi.begin(9600);

               ^

exit status 1

'wifi' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

**Additional info (please complete):**
 - MCU: [nodemcu ESP8266.]
 - Wireless module type [SX1278]
 - Arduino IDE version  [e.g. 1.8.13]

jgromes, 2021-02-08:
RadioLib ESP8266 driver is meant to be used when the ESP8266 is running AT firmware as a WiFi module connected to some other Arduino. To prevent collisions with the ESP8266 platform, this driver is disabled when RadioLib is running on ESP8266. So at the moment, RadioLib MQTT driver is not compatible with ESP8266 platform.

However, due to the modular nature of RadioLib, it should be possible to add that support by implementing `TransportLayer` interface using ESP8266 API.

nkebx2, 2021-02-19:
Noted. Thank you for your reply.

closed 2021-02-19 by nkebx2