jgromes/RadioLib #248: Random crashes when transmitting surely because of transmit collisions between two modules (ESP32)

ddrmax, 2021-02-12:
**Describe the bug**
Sometimes on ESP32 Boards, your system transmits data during the time another module is already transmitting, the ESP 32 crashes. in my minimal test sketch (only radio), I need to push the Core debug Level to see the backtrace, because at level none the ESP32 Hangs with no watchdog triggered, with level debug and up, the watchdog is triggered

here is the debug output

Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x40088d37  PS      : 0x00060b34  A0      : 0x4008a543  A1      : 0x3ffb1e40
A2      : 0x00050b23  A3      : 0x3ffc1190  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x03ad6678  A7      : 0x00000000  A8      : 0x80088d37  A9      : 0x3ffb1e20
A10     : 0x3ffbee04  A11     : 0x00000000  A12     : 0x3ffc1268  A13     : 0x3ffb8058
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000011  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x40088d37:0x3ffb1e40 0x4008a540:0x3ffb1e60 0x4008a5d1:0x3ffb1e80 0x400d1e17:0x3ffb1ea0 0x400d3631:0x3ffb1ec0 0x400d3a0b:0x3ffb1f00 0x400d3a28:0x3ffb1f20 0x400d1086:0x3ffb1f40 0x400d14cb:0x3ffb1f60 0x400d1943:0x3ffb1f90 0x400d5c21:0x3ffb1fb0 0x40088215:0x3ffb1fd0

Core 0 register dump:
PC      : 0x400ed2aa  PS      : 0x00060334  A0      : 0x800d7e62  A1      : 0x3ffbc070
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x00000001
A6      : 0x00060120  A7      : 0x00000000  A8      : 0x800d7a2a  A9      : 0x3ffbc040
A10     : 0x00000000  A11     : 0x40084d78  A12     : 0x00060120  A13     : 0x3ffbb6d0
A14     : 0x00000000  A15     : 0x3ffbbd60  SAR     : 0x00000000  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x400ed2aa:0x3ffbc070 0x400d7e5f:0x3ffbc090 0x40089706:0x3ffbc0b0 0x40088215:0x3ffbc0d0

Here what is the correspondance done by ESP Exception decoder:

PC: 0x40088d37: vTaskSwitchContext at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/include/freertos/portable.h line 208
EXCVADDR: 0x00000000

Decoding stack results
0x40088d37: vTaskSwitchContext at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/include/freertos/portable.h line 208
0x400d1e17: Module::yield() at C:\Arduino\portable\sketchbook\libraries\RadioLib\src\Module.cpp line 378
0x400d3631: SX127x::transmit(unsigned char*, unsigned int, unsigned char) at C:\Arduino\portable\sketchbook\libraries\RadioLib\src\modules\SX127x\SX127x.cpp line 157
0x400d3a0b: PhysicalLayer::transmit(char const*, unsigned char) at C:\Arduino\portable\sketchbook\libraries\RadioLib\src\protocols\PhysicalLayer\PhysicalLayer.cpp line 46
0x400d3a28: PhysicalLayer::transmit(String&, unsigned char) at C:\Arduino\portable\sketchbook\libraries\RadioLib\src\protocols\PhysicalLayer\PhysicalLayer.cpp line 42
0x400d1086: SendLoRaPacket(int, String, bool) at C:\Arduino\portable\sketchbook\DIGI_DEBUG_LORA/DIGI_DEBUG_LORA.ino line 112
0x400d14cb: Checkradio() at C:\Arduino\portable\sketchbook\DEBUG_LORA/DIGI_DEBUG_LORA.ino line 282
0x400d1943: loop() at C:\Arduino\portable\sketchbook\DEBUG_LORA/DIGI_DEBUG_LORA.ino line 415
0x400d5c21: loopTask(void*) at C:\Arduino\portable\packages\esp32\hardware\esp32\1.0.4\cores\esp32\main.cpp line 19
0x40088215: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

**To Reproduce**
no code for now
but a simple echo sketch on two esp32 modules will do the same problem

**Expected behavior**
No crashes but only packet Loss (since it's interfering)

**Additional info (please complete):**
 - MCU: ESP32 (Heltec LoRa V2)
 - Link to Arduino core: Latest [ESP 32](https://github.com/espressif/arduino-esp32)
 - Wireless module type [SX1276] tweaked for 868MHz
 - Arduino IDE version  [e.g. 1.8.13]
 - Library version [4.1.0 up to the latest 4.2.0]

jgromes, 2021-02-12:
I don't see how on-air collision could cause the ESP32 to crash, especially since it's crashing from a call to `yield()`.

I see from the backtrace you're running some custom sketch, does this happen when running the examples? From your description, just running two `SX127x_Transmit` examples on two ESP32 boards should be enough to crash them.

ddrmax, 2021-02-13:
Sorry for the long reply but here how I mostly resolved the problem:
I added the Channel activity detection, to block any potential collisions to my sketch.
But first, before posting I Recreated my sketch by only taking the radio part (Each modules have it's own ".h" files) where there's only the setup part (standard radiolib init for my board) and a receive and transmit loop based on the examples. I have done sorts of tests with and without serial.print to show if the data was valid (when serial.print is used the data is always valid)
The crash or worst, the module hanging is always repecuted by ESP exception decoder as coming from the SX127X.transmit()
I don't use Yield(); directly in my code dut I use some delays (a  random one before  retransmitting)
my smallest sketch crashing is a simple loop where the module is always in receive mode and when the data has specific header, It will put the data into stings and send to an other function, solely used for transmition that will add a new header and in the end transmit

ddrmax, 2021-02-13:
Only using channel activity detection seems to mitigate 3/4 of the crashes, because either protecting my transmit function with semaphores or critical sections doesn't help, my base code uses an Boolean called interrupt enable that is always set to false when it crashes so normally, the receive part of my sketch can't be called during the whole process of decoding and  transmitting

jgromes, 2021-02-13:
I'm very confused now.

Are you able to reproduce this crash with unmodified SX1278x_Transmit examples running on two ESP32 boards? If not, can you post minimal sketch and steps required to reproduce this?

ddrmax, 2021-02-13:
For a unknown reason I can't reproduce the problem for now, When I'm at the office, I can make it crash but now my basic sketch doesn't crashes when I flood it with messages (aka 20 nodes that have the CAD added into the firmware) that floods my test node by the time it is flooded, multiples nodes would have either reset or hang but here at my house, it's impossible to make it crash.

The test ran with no problem, the only problem is only due to the repeat mechanism that keeps repeating, like a ping pong match, the system ran for a good 10 minutes with a node bashing data to force the locked up state of indefinite repeat and 10 minutes of the repeat mechanism that kept the system flooded with the message before going back to normal.

The repeat problem will be solved by either adding a repeat counter and sequence counter in the data
I dunno if the crash could be due to another lora system in my victiny or the lorawan gateway near the office, but i use frequencies out of the lorawan channels that use another codding rate since I can have high rate of CRC errors (I use CRC) and I had a lot of -7 as a state value

jgromes, 2021-02-13:
OK I'm closing this now, until you can provide the sketch to reliably reproduce this behavior. Without it there's no way for me to help you, because from what you've posted so far, I have very little idea as to what is going on. What exactly were the debugging steps taken? What code are you running? What is the hardware setup?

Claiming "Random crashes" are "surely because of transmit collisions" seems rather premature here.

closed 2021-02-13 by jgromes