jgromes/RadioLib #249: Interrupts not working on STM32 Arduino core (default platform)

ahessling, 2021-02-12:
Hi!

I am trying to get the interrupt-driven packet receive mode to work, using a STM32G081RB controller in combination with the official STM32 Arduino core (https://github.com/stm32duino/Arduino_Core_STM32/, version 1.9.0) and a RFM95W module (SX1276).
The most basic sketch is your example "SX127x_Receive_Interrupt".

The problem is that no interrupts are generated, although packets are definitely received. The DIO0 line becomes high (as it should be) when the first packet is received. When I use the polling-mode (as in example SX127x_Receive), packets are successfully received.

Debugging the whole thing leads to the following conclusion:
- DIO0 in my case is mapped to digital pin 30, which is STM32 pin "PC8"
- When the function `setDio0Action` is called, `Module::attachInterrupt` is called with `RADIOLIB_DIGITAL_PIN_TO_INTERRUPT` with digital pin 30 as its argument (so far, so good)

Now here's the problem:
For `ARDUINO_ARCH_STM32`, this is defined in BuildOpt.h:
`
    #define RADIOLIB_DIGITAL_PIN_TO_INTERRUPT(p)        digitalPinToInterrupt(digitalPinToPinName(p))
`

The function `attachInterrupt` in file WInterrupts.cpp of the Arduino core is called with the result of the above macro, which says:
`
  PinName p = digitalPinToPinName(pin);
`
pin being the result of the macro.

So the already resolved pin name (30 --> PC8, which is defined to 40) is wrongly resolved a second time to a completly wrong pin.

I think the solution is to change the macro to:
`
    #define RADIOLIB_DIGITAL_PIN_TO_INTERRUPT(p)        digitalPinToInterrupt(p)
`

In my case, this works just perfect and I think that is the solution. Since I am new to RadioLib and the (STM32) Arduino core, I am not 100% sure of the implications, so I generated this issue. Please check if this change would be OK.

Kind regards,
André

jgromes, 2021-02-12:
That's odd, I'm quite certain I've had interrupts working on STM32L452RE in the past.

Which pin naming are you using? `PC8` or `PC_8`? The one without the underscore is just a macro that resolves to an integer, while the other one is its own datatype `PinName`, that could make a difference since I only ever used the first one.

Also, I can't find the variant files for STM32G081RB in the 1.9.0 STM32 core, is that something custom?

ahessling, 2021-02-12:
STM32G081RB is something custom, it is based on NUCLEO_G071RB with adjustments matching my custom hardware.

Concerning the pin naming, I am just using numbers (the digital pins).

This is the call of the constructor (_radioModule is of type Module):
`_radioModule(28/*cs*/, 30/*dio0*/, 29/*reset*/, 31/*dio1*/, _spi2, SPISettings(2000000, MSBFIRST, SPI_MODE0))`

And my variant.cpp looks like this (just an extract):
```
const PinName digitalPin[] = {
  PD_4,  //D0/SW1
  PD_3,  //D1/SW2
  PD_2,  //D2/SW3
...
  PB_10, //D25/SPI2_CLK
  PB_11, //D26/SPI2_MOSI
  PB_2,  //D27/SPI2_MISO
  PB_12, //D28/RFM_CS
  PB_13, //D29/RFM_RESET
  PC_8,  //D30/RFM_DIO0
  PC_9,  //D31/RFM_DIO1
};
```

So digital pin 30 really is PC_8 which is physically connected to the DIO0 pin of the module.

It just occurs odd to me that the macro RADIOLIB_DIGITAL_PIN_TO_INTERRUPT(p) resolves the pin number to the PinName (30 to PC_8 in my case), and the function attachInterrupt of the Arduino core does the same again, but with the already resolved PinName (PC_8, which is actually of type PinName and defined to 40 in PinNames.h). The function `attachInterrupt` will then resolve 40 to whatever is defined in the array `digitalPin` (which is PD_6 in my case and completly wrong).

If you consider the other platforms in BuildOpt.h, `RADIOLIB_DIGITAL_PIN_TO_INTERRUPT` is always defined to `digitalPinToInterrupt(p)`, except for ARDUINO_ARCH_STM32 (which is what I am using and only working, if I use the same define as for the other platforms).

jgromes, 2021-02-12:
I think I tracked down what the issue was - you're right, `RADIOLIB_DIGITAL_PIN_TO_INTERRUPT ` shouldn't call `digitalPinToPinName()`. STM32 interrupts worked for me previously since the `RADIOLIB_DIGITAL_PIN_TO_INTERRUPT` macro didn't exist in RadioLib when I last used it with STM32s (that was probably around 3.6.0). So really this was a case of improving something so much it breaks :)

On a side note, the fact that one of the macros differs from the others isn't really all that strange, in fact it's the very reason those macros exist. Different Arduino cores sometimes introduce subtle incompatibilities, this is a way to resolve them at compile time.

Fixed now, thanks for reporting!

closed 2021-02-12 by jgromes