jgromes/RadioLib #251: Potential wrong sequence at initializing SX1276 module.

HamzaHajeir, 2021-02-13:
Hi there

I've looked in SX1276 to see how the settings work, and what would be different to SX1278. beside reading the datasheet (didn't read it yet).

But I thought of a potential wrong sequence here : 

https://github.com/jgromes/RadioLib/blob/master/src/modules/SX127x/SX1276.cpp#L60

**What's wrong ?** 
1. `SX1276::begin()` starts calling `SX1278::begin()`, There's no problem in that.
2. Then it sets the frequency by calling `SX1276::setFrequency();`
3. `SX1276::setFrequency();` depends on the value of `_bw` _link above_, Which is not yet initialized. See [this line](https://github.com/jgromes/RadioLib/blob/master/src/modules/SX127x/SX1276.cpp#L17), Where `_bw`'s initial value is [zero](https://github.com/jgromes/RadioLib/blob/master/src/modules/SX127x/SX127x.h#L954).

Have I missed something ? 

Thanks

jgromes, 2021-02-13:
That's a very good catch, thanks. The part of the code you're referring to in `SX1276::setFrequency()` actually fixes an issue from SX127x errata sheet. So at the moment, since _bw is set to 0, that fix won't be executed unless you explicity call `setFrequency` again after `begin`. Will fix ASAP.

jgromes, 2021-02-13:
Should be fixeed now by swapping order of frequency and bandwidth configuration. Thanks for reporting!

closed 2021-02-13 by jgromes

HamzaHajeir, 2021-02-13:
Very quick reaction! 

Glad to contribute in some way. 

All the best ;)