jgromes/RadioLib #252: Custom SPI pins + Interrupt receive/transmit on ESP32 - NRF24L01

PacciK, 2021-02-14:
Hi. I have managed to get RadioLib to work on custom SPI pins, at least for the TX part, using [the two examples](https://github.com/jgromes/RadioLib/tree/master/examples/nRF24) provided as a base, and then adding the following for custom pins:

`  #define NRF24_CSN   0
  #define NRF24_IRQ   39 //  << This is a GPI ( not  usable as an output )
  #define NRF24_CE    26
  #define VSPI_MISO   22
  #define VSPI_MOSI   21
  #define VSPI_SCLK   2
  #define VSPI_SS     NRF24_CSN  //  NRF24_CE( I tried swapping these too - no joy)

SPIClass newSPI(VSPI);
nRF24 radio = new Module(NRF24_CSN, NRF24_IRQ, NRF24_CE, RADIOLIB_NC, newSPI);
`

Can we have an RX example working with Interrupt, @jgromes , since lines 67/68 say
`  //       See example ReceiveInterrupt for details
  //       on non-blocking reception method.`
but there is no Interrupt based example.

When I tried to adapt from other IRQ modules, I still couldn't get it to work well, resulting in some wine drinking to keep calm instead of bashing one's head :)

Cheers !

jgromes, 2021-02-14:
You're right, the interrupt examples are indeed missing - looks like I just forgot to add them, sorry about that. However, the logic should be the exact same as all the other modules: 

1. initialize the module with `begin()`,
2. call `setIrqAction()` to set the interrupt service routine (you can take the setFlag function from interrupt examples for other modules),
3. call `startReceive()` to put the module in listen mode,
4. in loop, call `readData()` when the interrupt flag is set

I'll add the interrupt examples as soon as I can, but there should be no reason for it to not work. I noticed you marked the IRQ as "GPI ( not usable as an output )", does your nRF24 not have an IRQ pin?

PacciK, 2021-02-14:
Hi @jgromes  Thanks for super quick reply !

>>>I'll add the interrupt examples as soon as I can, but there should be no reason for it to not work. I noticed you marked the IRQ as "GPI ( not usable as an output )", does your nRF24 not have an IRQ pin?<<<
Yes, of course it has an IRQ pin, and its connected to a GPI(not GPIO).  This is because pin 39 on an ESP32 works only as an Input, which suffices for an Input trigger by interrupt. [LINK to Espressif GPIO doc](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/gpio.html)

>>>However, the logic should be the exact same as all the other modules:<<<

Yes, I am doing exactly that:
`
#define NRF24_CSN 0
#define NRF24_IRQ 39 // << This is a GPI ( not usable as an output )
#define NRF24_CE 26
#define VSPI_MISO 22
#define VSPI_MOSI 21
#define VSPI_SCLK 2
#define VSPI_SS NRF24_CSN // NRF24_CE( I tried swapping these too - no joy)

SPIClass newSPI(VSPI);
nRF24 radio = new Module(NRF24_CSN, NRF24_IRQ, NRF24_CE, RADIOLIB_NC, newSPI);
`
Then, in Setup:
`newSPI.begin(VSPI_SCLK, VSPI_MISO, VSPI_MOSI, VSPI_SS);
  Serial.print(F("[nRF24] Initializing ... "));
  int state = radio.begin(2400, 1000, -18, 5); // Freq, DataRate, Power. AddressWidth
  if(state == ERR_NONE) {
    Serial.println(F("success!")); } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while(true);
  }

  Serial.print(F("[nRF24] Setting address for receive pipe 0 ... "));
  byte addr[] = {0x01, 0x23, 0x45, 0x67, 0x89};

  radio.setIrqAction(setFlag);

  state = radio.setReceivePipe(0, addr);
  if(state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while(true);
  }

  state = radio.startReceive();
  if(state == ERR_NONE) {
    Serial.println(F("Start RX success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while(true);
  }

enableInterrupt = true;

`

Then in loop function, I adapted from your other interrupt examples for other modules:

`
if(receivedFlag) {
  enableInterrupt = false;
  receivedFlag = false;

  String str;
  int state = radio.receive(str);    // << @jgromes  this produces data, but also a lot of repetitions on each Serial.Print
//int state = radio.readData(str); // << @jgromes - this produces no data !!
   

  // you can also receive data as byte array
  /*
    byte byteArr[8];
    int state = radio.receive(byteArr, 8);
  */

  if (state == ERR_NONE) {
    // packet was successfully received
    /////Serial.println(F("success!"));

    // print the data of the packet
    Serial.print(F("[nRF24] Data:\t\t"));
    Serial.println(str);
    Serial.flush();

  } else if (state == ERR_RX_TIMEOUT) {
    // timeout occurred while waiting for a packet
    Serial.println(F("timeout!"));

  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);
  }

    radio.startReceive();     // put module back to listen mode
    enableInterrupt = true;   // enable interrupt service routine
  }
`

For example, i am sending a 32 character payload. And this is what I receive:

`[nRF24] Data:		Hello World! 123456789ABCDEFGHI
[nRF24] Data:		IIIIIIIIIIIIIIIIIIIIIIIIIIIIIII
[nRF24] Data:		IIIIIIIIIIIIIIIIIIIIIIIIIIIIIII
[nRF24] Data:		IIIIIIIIIIIIIIIIIIIIIIIIIIIIIII
[nRF24] Data:		IIIIIIIIIIIIIIIIIIIIIIIIIIIIIII
[nRF24] Data:		IIIIIIIIIIIIIIIIIIIIIIIIIIIIIII
[nRF24] Data:		IIIIIIIIIIIIIIIIIIIIIIIIIIIIIII
[nRF24] Data:		IIIIIIIIIIIIIIIIIIIIIIIIIIIIIII
[nRF24] Data:		IIIIIIIIIIIIIIIIIIIIIIIIIIIIIII
[nRF24] Data:		IIIIIIIIIIIIIIIIIIIIIIIIIIIIIII
[nRF24] Data:		IIIIIIIIIIIIIIIIIIIIIIIIIIIIIII
[nRF24] Data:		IIIIIIIIIIIIIIIIIIIIIIIIIIIIIII
[nRF24] Data:		IIIIIIIIIIIIIIIIIIIIIIIIIIIIIII
[nRF24] Data:		IIIIIIIIIIIIIIIIIIIIIIIIIIIIIII
[nRF24] Data:		IIIIIIIIIIIIIIIIIIIIIIIIIIIIIII
[nRF24] Data:		IIIIIIIIIIIIIIIIIIIIIIIIIIIIIII
[nRF24] Data:		IIIIIIIIIIIIIIIIIIIIIIIIIIIIIII
[nRF24] Data:		Hello World! 123456789ABCDEFGHI`

As you can see the last character is repeating until fresh data is coming.

I also suspect that the irq is not firing, as I never see the "I have an interrupt" in the setFlag Function I modified to check for irq action....

`void setFlag(void) {

  Serial.println(F("I have an Interrupt!"));
  
  if(!enableInterrupt) {    // check if the interrupt is enabled
    return;
  }
  receivedFlag = true;      // we got a packet, set the flag
}`

PacciK, 2021-02-14:
PS.. apologies but the "Insert Code" button is not working well for me.

jgromes, 2021-02-15:
Couple of things I noticed:

1. You can't `Serial.println()` from an interrupt service routine - as long as you are servicing the current interrupt, no more interrupts can be serviced, so nothing that uses interrupts (Serial, delay, millis, etc.) will work.

2. `receive()` is a blocking method, you have to use `readData()` here.

Remove the Serial.print from setFlag and try with `readData()`. If nothing is printed, then that means the interrupt indeed isn't firing and we'll debug further.

PacciK, 2021-02-15:
@jgromes Thanks for your suggestions.

I built a breadboard version, then connected my Oscilloscope to the IRQ pin, and started changing the assignment of the IRQ pin around until RX worked ( and the IRQ pin started going low ).
And I noticed that somehow it would only work if I had my finger on a couple of pins....
Strange, I thought...  hmm Impedance ?? Not in the digital world.....
On 'straigntforward' GPIO's the interrupt fires without issue.

Well, after a lot more prodding about... the solution was to declare ( this special GPI ) IRQ pin as an **input**, in the **Setup()**.
``
pinMode( NRF24_IRQ, INPUT );
``

closed 2021-02-15 by PacciK

jgromes, 2021-02-15:
Good to know you resolved it - this is actually supposed to be done in the begin() method. All the other modules do it, but for some reason it was missing from nRF24. It wasn't an issue before because almost all MCUs I'm aware of have input pins by default.

Should be fixed now. On a side note it's a bit funny you have to explicitly set mode to input on input-only pin to make it work.

PacciK, 2021-02-15:
>On a side note it's a bit funny you have to explicitly set mode to input on input-only pin to make it work.

True !
The hint was the "hmm Impedance ??"