jgromes/RadioLib #253: Not

HamzaHajeir, 2021-02-17:
Hi

Investigating (again) in the difference between SX1278 and SX1276 implementations (please tell me).  to mimic the change to the other library that's used in STM32CubeIDE.

I saw something that could go wrong : 
[SX1276.cpp :: 120](https://github.com/jgromes/RadioLib/blob/master/src/modules/SX127x/SX1276.cpp#L120).

It only returns the state of setFrequencyRaw();
while the code in [SX1278.cpp](https://github.com/jgromes/RadioLib/blob/master/src/modules/SX127x/SX1278.cpp#L132) updates the class field _freq with new one.

I think it's a wrong behavior since the parent's [SX1278::getRSSI()](https://github.com/jgromes/RadioLib/blob/master/src/modules/SX127x/SX1278.cpp#L405) checks for the frequency, While SX1278 operates on a lower frequencies, SX1276 operates on higher frequencies too.

Is that missed ?

Thanks

jgromes, 2021-02-18:
Yet another good catch, many thanks! You're right, that is indeed missing, for all classes derived from SX1278 in fact. Will fix ASAP.

jgromes, 2021-02-18:
Should be fixed now for SX1276/77/79 as well as RFM95/96, thanks for reporting!

closed 2021-02-18 by jgromes

HamzaHajeir, 2021-02-18:
You're welcomed ;)