jgromes/RadioLib #254: Waterfall visible callsign/ID?

microfibre-cpc, 2021-02-18:
Might it be technically possible to implement something along the lines of the waterfall-visible callsign/ID as is found for some PC-based SSTV transmissions? e.g. 

[https://community.flexradio.com/discussion/6493264/how-does-one-get-their-callsign-to-appear-in-the-waterfall](https://community.flexradio.com/discussion/6493264/how-does-one-get-their-callsign-to-appear-in-the-waterfall)

I wouldn't know where to begin to work out the required bandwidth and whether any of the supported modules could be shoehorned to perform this in fsk/afsk.

jgromes, 2021-02-18:
One step ahead of you :P

https://twitter.com/JanGromes/status/1248518019862667264/photo/1

So yeah, it's absolutely possible, but I'm not sure I want to release the sketch to do that into the wild. The signal in that photo took up about 1 MHz of bandwidth.

microfibre-cpc, 2021-02-18:
Neat!  

You could limit the bandwidth to ~3kHz to match SSTV.? Vertical scroll rather than 'landscape'...glorified adaptation of the hellschreiber mode!

A little more research unearths a few utilities to generate a sound 'picture'

[https://www.skytopia.com/software/sonicphoto/index.htm
](https://www.skytopia.com/software/sonicphoto/index.htm)
[https://eax.me/waterfall-images/](https://eax.me/waterfall-images/)

Thinking out loud, the .wav data could be compiled off-chip; the raw sample data then uploaded into memory on the host board instead of being calculated on the fly, which could also constrain the BW. This is then 'sent' via the (A)FSK on the chip.

It would be great if you could release even a gist of the code, just because you have such a good head start on the rest of us!

jgromes, 2021-02-18:
I do love myself some Hellschreiber, however, I don't think you'll be able to get very good resolution if your bandwidth limit is 3 kHz. Most common modules (SX127x, RF69 etc.) have a synthesizer resolution of a couple dozen Hz.

Also, it's been nearly a year since that Twitter post - I don't actually have the codes for that anymore :D

But I do recall it was implemented in an Arduino sketch, no modification to the library was needed - just some calls to `setFreqency()` and `transmitDirect()`, basically turning on the carrier and modulating it from software.

closed 2021-02-18 by jgromes