jgromes/RadioLib #257: Support for Heltec CubeCell boards

spleenware, 2021-02-22:
I am experimenting with the HelTec CubeCell boards, which have Arduino support.
The RadioLib library doesn't compile, so I assume it isn't supported when targeting these boards. 
(They use the sx1262 radio module.)

I am initialising in my sketch like so:

#define  P_LORA_SCLK  P4_2
#define  P_LORA_MISO  P4_1
#define  P_LORA_MOSI  P4_0

#define  P_LORA_DIO_1  P4_6
#define  P_LORA_BUSY   P4_7
#define  P_LORA_NSS    P4_3
#define  P_LORA_RESET  P5_7

static SX1262 radio = new Module(P_LORA_NSS, P_LORA_DIO_1, P_LORA_RESET, P_LORA_BUSY);

jgromes, 2021-02-22:
Well since CubeCell isn't mentioned in the list of supported platforms in readme, I would expect it to be, well ... not supported.

Can you post basic information, such as a link to the Arduino core for CubeCell? I've never heard of it.

jgromes, 2021-02-22:
Nevermind, I did actually hear of it - in a previous issue: https://github.com/jgromes/RadioLib/issues/187
From that, it's clear that CubeCell can't be supported unless its broken Arduino compatiblity is fixed.

closed 2021-02-22 by jgromes

spleenware, 2021-02-23:
Sorry, didn't see that this had already been raised. Is a shame Heltec did such a hatchet job on the Arduino support :-(
The CubeCell boards seem to be very good at low power mode, and has solar charging, so was hoping to do long-running experiments.