jgromes/RadioLib #260: PingPong example request for SX126x modules

PacciK, 2021-02-25:
Hello @jgromes 
I am experimenting with two Ebyte E22-900M30S modules which use the SX1262 chip + PA, on the **ESP32** platform
I have tried the examples for this module and they work well.
Now I am looking at building a **fully interrupt** based half duplex system, but am struggling.

I've had a look at this [discussion](https://github.com/jgromes/RadioLib/discussions/237) where the author describes he has built a working full interrupt based on an ESP8266 + SX1278 scenario.

Translating this to the SX1262, it seems the SX126x library does not have the "getIRQFlags()" function but a seemingly similar function "getIrqStatus()" , which is a private function.
Therefore, I've (cautiously) enabled "GODMODE" to allow me to use "getIrqStatus()" in the example mentioned in the discussion post above.

However Arduino IDE is complaining and not even compiling the code because the compiler is complaining about enableInterrupt  not being declared, in the "enableInterrupt = true;" line, which is found towards the middle part of the loop() structure.

To cut a long story short, it would be super nice if You could add a fully interrupt based, ping-pong example, where one module sends a counter update, and the other module receives it, adds one value to the counter, and sends it back to the other module, starting a chain. This example would serve a multitude of teaching / testing purposes .

Thanks Jan !!

PS. on the Ebyte module I am using a mosfet tied to DIO2 to invert the drive for the RX switch, which Idea I got from another post referring to what seems like a nice trick circulating on the FossaSat forum- [LINK](https://meshtastic.discourse.group/t/new-hardware-prototype/133/58)

Thanks again for such a wonderful library !

jgromes, 2021-02-25:
> add a fully interrupt based, ping-pong example

I'll see if I can add that, not sure when will I get to it, but seems it would be useful.

> However Arduino IDE is not even compiling the code

I haven't tried to run that sketch, but you're right that `enableInterrupt` seems undefined in there.

> on the Ebyte module I am using a mosfet tied to DIO2 to invert the drive for the RX switch

I have a design that does the same using two NPNs, though the second one just drives an LED to clearly show the antenna switch status: https://github.com/jgromes/ESP32-1W-GroundStation/blob/master/doc/sch.pdf Also the link points to Meshtastic forums, not to FOSSA, I'm not sure if that's what you meant to post.

PacciK, 2021-02-25:
Hi Jan ! 
Thanks for your quick reply and for the link to your GroundStation project. 

Yes my link points to a meshtastic discussion, which then references to a schematic ( around 9 posts up from the bottom ) which then references a FOSSA forum ( which I've tried to search for, but perhaps Mr.Google does not know about....)

PacciK, 2021-02-25:
[image]

jgromes, 2021-02-26:
It's probably from the Telegram channel then, or rahter what used to be FOSSASAT-1 Community telegram and is now the tiny.gs project.

PacciK, 2021-03-11:
any update, @jgromes 
Thank You !

jgromes, 2021-03-12:
Sorry, there's been zero progress on this on my end, I've been unfortunately quite busy.

jgromes, 2021-06-17:
Example added in recent commit.

closed 2021-06-17 by jgromes