jgromes/RadioLib #265: GMSK between two sx1278

yeckel, 2021-03-13:
Hi, pls I'm trying to TX D-Star DV messages from ttgo t-beam (esp32, GPS, sx1278 433MHz), however, I've got to the point when I'm unable to RX&TX  GMSK packets between two SX1278 on my desk (not speaking to Icom trx). 

Could please someone point me to what I'm doing wrong?

[gmsk_rx.cpp](https://tomsik.eu/index.php/s/T6Se5KTulLXERmW)
[gmsk_tx.cpp](https://tomsik.eu/index.php/s/PK4EmEaO2xdeXLi)

```
 radio.reset();
    Serial.print(F("[SX1278] Initializing ... "));
    checkLoraState(radio.beginFSK(434.800f, 4.8f, 4.8 * 0.25f, 25.0f, 10, 32, false)); //6.25kHz bandwidth
    checkLoraState(radio.setSyncWord(syncWord, 4));
    checkLoraState(radio.setEncoding(RADIOLIB_ENCODING_NRZ));
    checkLoraState(radio.setDataShaping(RADIOLIB_SHAPING_0_5));
    radio.setCRC(false);
```

The modules are sending something that looks like gmsk, when I TX from IC-705 I can receive always the same packet on the board. Frequencies seem to be more/less correct. 
TNX de OK1CHP

jgromes, 2021-03-13:
> I'm unable to RX&TX GMSK packets between two SX1278 on my desk

Please describe the issue in a bit more detail - what exactly is the output on the receiver, no output, CRC errors, bad data ...? Are there any errors on the transmitter side?

Also, I noticed you're using blocking receive, interrupt-driven reception might really is the better approach here.

yeckel, 2021-03-13:
No RX, no error neither on TX nor RX side. I can see something being sent on Icom radio, also RX works when I press PTT in the D-Star mode. Just the two SX1278 don't speak with each other. One seems to transmit with ~2.5kHz offset, but I've also tried modifying the frequency for it. I've used interrupt-driven RX as well, blocking is just for strapping the code.

jgromes, 2021-03-13:
The frequency offset is not unexpected, the crystals on off-the-shelf SX127x modules are usually pretty bad.

The fact that it is still able to receive IC-705 messages is quite interesting - I suppose there could be some configuration issue on the transmitter side. One way to debug this could be to start removing bits of configuration until the two SX1278 can talk to each other again, I'd start with the shaping parameter.

yeckel, 2021-03-13:
Great tip with crystals! I've tried with LoRa and logged the frequency offset. Let it set for a while. The link was drifting heavily up to -1..+4kHz! And finally set the measured difference as an offset to the beginFSK() frequency (+3.5kHz). Now it receives almost every packet. When I was doing frequency sweep it apparently lacked some time to stabilize. TNX I was not aware of how bad those crystals are.

closed 2021-03-13 by yeckel

jgromes, 2021-03-13:
They're really quite something - I suggest trying SX126x with TCXO instead, those are usually a lot better.