jgromes/RadioLib #266: Request to add function to change CRC mode

ssshock, 2021-03-14:
I'd like to request a function to be able to change the CRC calculation type for SX127x. Or at least the SX1276.

The existing "SetCRC" function is in the SX1278 specific library, so maybe it should go there. Apologies for the formatting of the code below. I'm not sure of the correct method. The following could be inserted into SX1278.cpp

``
int16_t SX1278::setCRCMode(uint8_t crcMode) {
  if(getActiveModem() != SX127X_FSK_OOK) {
    return(ERR_WRONG_MODEM);
  }
    // set FSK CRC
    if(crcMode) {
      return(_mod->SPIsetRegValue(SX127X_REG_PACKET_CONFIG_1, SX127X_CRC_WHITENING_TYPE_IBM, 0, 0));
    } else {
      return(_mod->SPIsetRegValue(SX127X_REG_PACKET_CONFIG_1, SX127X_CRC_WHITENING_TYPE_CCITT, 0, 0));
    }
  }``

It does a check to make sure the modem is in FSK mode first, as the CRC config register is only applicable to FSK mode. 
The values to enter for crcMode are defined in the SX127x.h file. Bit 0 in register 0x30 (SX127X_REG_PACKET_CONFIG_1). Default is 0x0 or SX127X_CRC_WHITENING_TYPE_CCITT. Optional mode is 0x01 or SX127X_CRC_WHITENING_TYPE_IBM.

SX1278.h only needs the function definition.
``
/*!
      \brief Sets the CRC mode for FSK modem.

      \param Set crcMode to SX127X_CRC_WHITENING_TYPE_CCITT for CCITT(default)(X16 + X12 + X5 + 1) or SX127X_CRC_WHITENING_TYPE_IBM for IBM(X16 + X15 + X2 + 1).

      \returns \ref status_codes
    */
    int16_t setCRCMode(uint8_t crcMode);
``
Thanks.

ssshock, 2021-03-14:
For reference, this this setting that could be used by @G4lile0 in the TinyGS project to support other satellites, such as Lucky-7.

jgromes, 2021-03-14:
I think this can be added, but it might be better to extend the current `setCRC` method, similar to what SX126x has. So the new method would look like `setCRC(enable, mode)` where mode could only be applied in FSK for the reasons mentioned in the first comment.

jgromes, 2021-03-14:
Added in the latest commit.

closed 2021-03-14 by jgromes