jgromes/RadioLib #267: Issue with an RFM95 having the wrong address

Eugene-Levinson, 2021-03-14:
Hello, I am having an issue with an "RFM95" module. I originally looked to me that the issue is identical to https://github.com/jgromes/RadioLib/issues/147

I also have the module from Aliexpress, it is also labelled as RFM95 at 868Mhz but the actual chip on it is labelled RF96.
My original issue was that it was saying the module did not support the right frequency, but after trying different example codes I got it to give me the -16 error code. I applied the solution from #147 but it still gives me the same error.

Does anyone have any ideas of what else I can try?
Thank you in advance!

**Sketch that is causing the module fail**

```c++

/*
   RadioLib SX127x Transmit Example

   This example transmits packets using SX1278 LoRa radio module.
   Each packet contains up to 256 bytes of data, in the form of:
    - Arduino String
    - null-terminated char array (C-string)
    - arbitrary binary data (byte array)

   Other modules from SX127x/RFM9x family can also be used.

   For default module settings, see the wiki page
   https://github.com/jgromes/RadioLib/wiki/Default-configuration#sx127xrfm9x---lora-modem

   For full API reference, see the GitHub Pages
   https://jgromes.github.io/RadioLib/
*/

// include the library
#include <RadioLib.h>

// SX1278 has the following connections:
// NSS pin:   1
// DIO0 pin:  14
// RESET pin: 24
// DIO1 pin:  13
RFM95 radio = new Module(1, 14, 24, 13);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1278 radio = RadioShield.ModuleA;

void setup() {
  Serial.begin(9600);

  pinMode(10, OUTPUT);  //on my board I have pin 10 turn on the power to the Lora chip via a mosfet
  digitalWrite(10, HIGH);

  delay(5000);
  Serial.println(F("Started!"));

  // initialize SX1278 with default settings
  Serial.print(F("[SX1278] Initializing ... "));
  int state = radio.begin (868.0, 125.0, 7, 5, 0x12);
  
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  // some modules have an external RF switch
  // controlled via two pins (RX enable, TX enable)
  // to enable automatic control of the switch,
  // call the following method
  // RX enable:   4
  // TX enable:   5
  /*
    radio.setRfSwitchPins(4, 5);
  */
}

void loop() {
  digitalWrite(10, HIGH);
  Serial.print(F("[SX1278] Transmitting packet ... "));

  // you can transmit C-string or Arduino string up to
  // 256 characters long
  // NOTE: transmit() is a blocking method!
  //       See example SX127x_Transmit_Interrupt for details
  //       on non-blocking transmission method.
  int state = radio.transmit("Hello World!");

  // you can also transmit byte array up to 256 bytes long
  /*
    byte byteArr[] = {0x01, 0x23, 0x45, 0x67, 0x89, 0xAB, 0xCD, 0xEF};
    int state = radio.transmit(byteArr, 8);
  */

  if (state == ERR_NONE) {
    // the packet was successfully transmitted
    Serial.println(F(" success!"));

    // print measured data rate
    Serial.print(F("[SX1278] Datarate:\t"));
    Serial.print(radio.getDataRate());
    Serial.println(F(" bps"));

  } else if (state == ERR_PACKET_TOO_LONG) {
    // the supplied packet was longer than 256 bytes
    Serial.println(F("too long!"));

  } else if (state == ERR_TX_TIMEOUT) {
    // timeout occurred while transmitting packet
    Serial.println(F("timeout!"));

  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }

  // wait for a second before transmitting again
  delay(1000);
}
```

**Debug mode output**
18:40:58.961 -> failed, code -16
18:40:59.885 -> [SX1278] Transmitting packet ... R	1	88	
18:40:59.927 -> W	1	89	
18:40:59.966 -> R	1	89	
18:40:59.966 -> R	1	89	
18:40:59.966 -> R	1D	72	
18:40:59.966 -> R	1E	70	
18:41:00.005 -> R	20	0	
18:41:00.005 -> R	21	8	
18:41:00.005 -> R	1	89	
18:41:00.005 -> W	1	89	
18:41:00.005 -> R	1	89	
18:41:00.046 -> R	1	89	
18:41:00.046 -> R	40	40	
18:41:00.046 -> W	40	40	
18:41:00.085 -> R	40	40	
18:41:00.085 -> R	1	89	
18:41:00.085 -> W	12	FF	
18:41:00.085 -> R	22	C	
18:41:00.085 -> W	22	C	
18:41:00.126 -> R	22	C	
18:41:00.126 -> R	E	0	
18:41:00.126 -> W	E	0	
18:41:00.165 -> R	E	0	
18:41:00.165 -> R	D	C	
18:41:00.165 -> W	D	0	
18:41:00.165 -> R	D	0	
18:41:00.165 -> W	0	48	65	6C	6C	6F	20	57	6F	72	6C	64	21	
18:41:00.207 -> R	1	89	
18:41:00.245 -> W	1	8B	
18:41:00.245 -> R	1	88	
18:41:00.245 -> 
18:41:00.245 -> address:	0x1
18:41:00.245 -> bits:		2 0
18:41:00.286 -> value:		0b11
18:41:00.286 -> current:	0b10001001
18:41:00.326 -> mask:		0b111
18:41:00.326 -> new:		0b10001011
18:41:00.326 -> read:		0b10001000

**Additional info **
 - MCU: Atmega1284p
 - Link to Arduino core: https://github.com/MCUdude/MightyCore
 - Wireless module type Hope RF RFM95 868Mhz 
 - Arduino IDE version  1.8.10
 - Library version 4.3.0

jgromes, 2021-03-14:
In #147 all known RFM9x device IDs were added as valid. RadioLib should now recognize any RFM9x, so that doesn't seem related. The debug output also supports this - there's no issues communicating with the module until we attempt to transmit, at which point the module goes to sleep instead of transmitting.

One of the comments mentions you have a power switch for the LoRa module - what's the power source, can it provide enough current to supply the RFM96 when it starts transmitting?

Eugene-Levinson, 2021-03-14:
Thanks a lot for your quick response!
I am powering it from an FTDI breakout board, thought a small mosftet. It should be supplying enough current, but I will bypass it and see if that makes a difference.

Eugene-Levinson, 2021-03-14:
That did solve the issue! Seems I need a different mosfet. Thanks a lot for your help!

closed 2021-03-14 by Eugene-Levinson

jgromes, 2021-03-14:
Glad you got it working - really goes to show how useful is the debug mode sometimes :)