jgromes/RadioLib #269: First byte missing in FSK

yeckel, 2021-03-15:
Hi Honza,
I'm trying with D-Star over sx1278 and struggling all over the way. It seems to me FSK is always missing the first byte.
Receive data: 0xB3,0xF7,0x32....
Sent data: **0x1c**, 0xb3, 0xf7, 0x32...
I was trying with shortening the sync word:
```uint8_t syncWord[] = {0x55, 0x55, 0x76, 0x50};```
to 
```uint8_t syncWord[] = {0x55, 0x55, 0x76};```
But then I'm unable to receive anything.

Could it be that there is some FSK issue in the sx1278? The LoRa is working perfectly.

Code used:
```
    checkLoraState(radio.beginFSK(f, 4.8f, 4.8 * 0.25f, 25.0f, 17, 0, false));
    checkLoraState(radio.setSyncWord(syncWord, sizeof(syncWord)));
    checkLoraState(radio.setEncoding(RADIOLIB_ENCODING_NRZ));
    checkLoraState(radio.setDataShaping(RADIOLIB_SHAPING_0_5));
```
[gmsk_receive.cpp](https://tomsik.eu/index.php/s/GgDvfRN05C34k3z)

I was checking the RadioLib code ```int16_t SX127x::readData(uint8_t* data, size_t len)``` and all seems to be fine, no skipping of the first byte there. 

Any ideas? TNX

jgromes, 2021-03-15:
Which packet mode are you using - variable or fixed length? And is the same on both the transmitter and the receiver?

yeckel, 2021-03-16:
Packet mode with a fixed length. and the TX is IC-705 from Icom. Now I've turned to direct mode and it's much better, taking bits from DIO2&DIO1. Could it be that the first byte in PSK mode is used as PKT length? Like in LoRa when ImplicitHeaderModeOn? When I switch from ```beginFSK()``` to LoRa ```begin()``` then the whole packet is received. Also data transfer routine is probably correct.

yeckel, 2021-03-16:
I found it, in the SX127x Datasheet chapter 4.2.13.2.  Packet Format the "Variable Length Packet Format" is the first byte used as a packet length. Also, I have to probably use "Unlimited Length Packet Format" set with ```fixedPacketLengthMode(0);```

jgromes, 2021-03-17:
Yes, that's the difference between fixed and variable packet length mode - in variable length mode, first byte after sync word is considered to be the length of the packet. In fixed length mode, the packet length must be known in advance, and the length byte is not included in the packet. What is the packet format that IC-705 is using? Preamble-sync-payload, or are there some other fields?

Also, if you have a sketch that's using direct mode on DIO1/2 to receive the data as a stream, that would be interesting and something I wanted to add to the library for quite a while now.

yeckel, 2021-03-17:
[D-Star](https://www.jarl.com/d-star/shogen.pdf) has: preamble + sync + header + Nx(voice frame 72byte, data frame 24byte) + stop-frame 24byte. Also "Unlimited Length Packet Format" or direct mode. TX with DIO1/2 works like a charm. RX is worse. I'll try coding it on the weekend.

yeckel, 2021-03-19:
Hi Honza, I've implemented basic GMSK [D-Star receiver](https://github.com/yeckel/D-StarReceive) Feel free to use any part of it, or suggest me in what form you wanna add it into the library so I can prepare it.

closed 2021-03-19 by yeckel

jgromes, 2021-03-22:
Thanks, this is going to be really useful. For some time I wanted to add some form of direct mode support, since there's a couple of use-cases that would benefit from it - guess I'll finally have to figure out some nice API :)