jgromes/RadioLib #270: SX127x_FSK_Modem.ino : for FSK & OOK enable data whitening by default

LieBtrau, 2021-03-15:
Hi Jan,

As you might have noticed, I've been playing around with your library the past couple of days.  Your library generally works very well.
For inexperienced users, the amount of settings a SX127x radio offers, might be overwhelming.  The examples can help the users to start with settings that work well and move on from there.  In that respect, I can recommend the RadioHead library approach.  It has a LUT (just an enum) that lists radio configs.  The user selects a radio config from that enumeration.  The library then configures frequency, bandwidth, baudrate according to the selected LUT value. e.g. [RH_RF95.h, ModemConfigChoice on line 595](https://github.com/adafruit/RadioHead/blob/master/RH_RF95.h).

For this SX127x_FSK_Modem.ino example specifically, I would enable data whitening by default (`setEncoding(RADIOLIB_ENCODING_WHITENING`).  It really makes a huge impact on receiver performance, especially if you were trying to send long strings of zeroes like I was doing.

Regards,

Christoph

jgromes, 2021-03-15:
I'm aware whitening will have significant impact on performace. Actually, there was a point during the development when it was enabled by default. However, this was later changed to NRZ because it had two consequences:

1. It "quietly" changes the transmitted data in some way the user might not be expecting - when you call `radio.transmit("abcd");` with whitening enabled, you would expect to see the "abcd" string on-air. With whitening enabled, you won't actually see the "abcd" string on-air, just the result of the whitening.

2. Not all FSK modules supported by RadioLib support whitening with the same polynomial. The logic was that by default, when not supplying any arguments to `radio.begin()`, transmissions from all FSK modules should be compatible, or at least have similar performance.

I'm also not entirely sure we need some configuration presets, at least not within the library code. It could however be argued that this should be handled by documentation, for example by a Wiki page that lists all parameters in the library and briefly explains them (with links to further reading). Overall, I don't think the library should make decisions for the user, it's up to them to pick the configuration that best fits their use-case.

LieBtrau, 2021-03-16:
It's indeed better to turn data whitening off to ensure interoperability with devices from other manufacturers and guarantee compatibility with ham-radio protocols.

I understand your point that a library shouldn't enforce a configuration upon the user.  Documentation could take care of that.  
But...every hour can only be spent once: you either spend it on writing documentation or on adding extra devices/features to the library.
For me personally, such documentation would be a nice to have, but I'll probably read the transceiver's datasheet anyway to learn about its features.
I'm glad that you respond quickly when (minor) bugs are found.  Thanks for the support.

closed 2021-03-16 by LieBtrau