jgromes/RadioLib #271: SX126x_Receive_Interrupt STM32 Nucleo-32

JoseCSNeto, 2021-03-16:
I was able to communicate between two Arduinos using your library/examples. Thank you for that.

Now I'm trying to communicate between an Arduino and a NUCLEO-G431KB. Tests using NUCLEO-G431KB as the transmitter and the Arduino as the receiver were successful, but I can't do it the other way around. The DIO1 never generates an interrupt that signals a message reception on the NUCLEO-G431KB. 

During the tests with NUCLEO-G431KB as the transmitter and the Arduino as the receiver I had to force the _transmittedFlag_ to _true_ in order to be able to do it successfully. But only need to do it after the first transmission. After that, the sketch runs without any problem. Maybe the problem with the receiver is related to this?

```c++   
transmissionState = radio.startTransmit("Hello World!");
transmittedFlag = true;
```

The tests mentioned above are based on the Interrupt examples. Tried testing with the blocking functions, but timeouts occur during reception. 

**Sketch that is causing the module fail**

```c++
// include the library
#include <RadioLib.h>

// SX1262 has the following connections:
#define CS PA4
#define DIO1 PB0
#define RESET PA0
#define BUSY PB4

#define MOSI PA7
#define MISO PA6
#define SCLK PA5

SX1262 radio = new Module(CS, DIO1, RESET, BUSY);

#define TX PA3
#define RX PA11

// flag to indicate that a packet was received
volatile bool receivedFlag = false;

// disable interrupt when it's not needed
volatile bool enableInterrupt = true;

void setFlag(void);

void setup() {
  delay(3000);
  Serial.begin(9600);
  SPI.setMISO(MISO);
  SPI.setMOSI(MOSI);
  SPI.setSCLK(SCLK);
  SPI.setSSEL(CS);
  SPI.begin(1);

  // initialize SX1262 with default settings
  Serial.print(F("[SX1262] Initializing ... "));
    int state = radio.begin(868.0, 125.0, 9, 7, SX126X_SYNC_WORD_PRIVATE, 22, 8, 0, true);
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
  radio.setRfSwitchPins(RX, TX);
  //pinMode(DIO1, INPUT);
  //attachInterrupt(DIO1, setFlag, RISING);

  // set the function that will be called
  // when new packet is received
  radio.setDio1Action(setFlag);

  // start listening for LoRa packets
  Serial.print(F("[SX1262] Starting to listen ... "));
  state = radio.startReceive();
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
}

// this function is called when a complete packet
// is received by the module
// IMPORTANT: this function MUST be 'void' type
//            and MUST NOT have any arguments!
void setFlag(void) {
  // check if the interrupt is enabled
  if(!enableInterrupt) {
    return;
  }

  // we got a packet, set the flag
  receivedFlag = true;
}

void loop() {
  // check if the flag is set
  if(receivedFlag) {
    // disable the interrupt service routine while
    // processing the data
    enableInterrupt = false;

    // reset flag
    receivedFlag = false;

    // you can read received data as an Arduino String
    String str;
    int state = radio.readData(str);

    // you can also read received data as byte array
    
     // byte byteArr[8];
     // int state = radio.readData(byteArr, 8);
    

    if (state == ERR_NONE) {
      // packet was successfully received
      Serial.println(F("[SX1262] Received packet!"));

      // print data of the packet
      Serial.print(F("[SX1262] Data:\t\t"));
      /*
      for (int i = 0; i< sizeof(byteArr);i++)
        Serial.print (byteArr[i]);
      */
      Serial.println(str);

      // print RSSI (Received Signal Strength Indicator)
      Serial.print(F("[SX1262] RSSI:\t\t"));
      Serial.print(radio.getRSSI());
      Serial.println(F(" dBm"));

      // print SNR (Signal-to-Noise Ratio)
      Serial.print(F("[SX1262] SNR:\t\t"));
      Serial.print(radio.getSNR());
      Serial.println(F(" dB"));

    } else if (state == ERR_CRC_MISMATCH) {
      // packet was received, but is malformed
      Serial.println(F("CRC error!"));

    } else {
      // some other error occurred
      Serial.print(F("failed, code "));
      Serial.println(state);

    }

    // put module back to listen mode
    radio.startReceive();

    // we're ready to receive more packets,
    // enable interrupt service routine
    enableInterrupt = true;
  }

}
```

**Hardware setup**
[image]

**Debug mode output**
```c++
[SX1262] Initializing ... M.SX126x
CMD.80.
DATW.0.A2.

CMD.80.
DATW.0.A2.

CMD.8F.
DATW.0.A2.0.A2.

CMD.8A.
DATW.1.A2.

CMD.93.
DATW.20.A2.

CMD.88.
DATW.3.A2.16.A2.A.A2.0.A2.0.A2.0.A2.0.A2.

CMD.2.
DATW.3.A2.FF.A2.

CMD.8.
DATW.0.A2.0.A2.0.A2.0.A2.0.A2.0.A2.0.A2.0.A2.

CMD.89.
DATW.7F.A2.

CMD.11.
DATR.0.A2.0.1.

Symbol length: 4.10 ms
CMD.8B.
DATW.9.A2.4.A2.3.A2.0.A2.

CMD.11.
DATR.0.A2.0.1.

Symbol length: 4.10 ms
CMD.8B.
DATW.9.A2.4.A2.3.A2.0.A2.

CMD.11.
DATR.0.A2.0.1.

Symbol length: 4.10 ms
CMD.8B.
DATW.9.A2.4.A2.3.A2.0.A2.

CMD.11.
DATR.0.A2.0.1.

CMD.D.7.40.
DATW.14.A2.24.A2.

CMD.11.
DATR.0.A2.0.1.

CMD.1D.7.36.
DATR.0.A2.0.D.

CMD.D.7.36.
DATW.9.A2.

CMD.8C.
DATW.0.A2.8.A2.0.A2.FF.A2.1.A2.0.A2.

CMD.D.8.E7.
DATW.18.A2.

CMD.9D.
DATW.1.A2.

CMD.96.
DATW.0.A2.

CMD.98.
DATW.D7.A2.DB.A2.

CMD.86.
DATW.36.A2.40.A2.0.B2.0.B2.

CMD.1D.8.E7.
DATR.0.A2.0.18.

CMD.95.
DATW.4.A2.7.A2.0.A2.1.A2.

CMD.8E.
DATW.16.A2.4.A2.

CMD.D.8.E7.
DATW.18.A2.

CMD.1D.8.D8.
DATR.0.A2.0.E8.

CMD.D.8.D8.
DATW.FE.A2.

success!
[SX1262] Starting to listen ... CMD.8.
DATW.2.A2.62.A2.0.A2.2.A2.0.A2.0.A2.0.A2.0.A2.

CMD.8F.
DATW.0.A2.0.A2.

CMD.2.
DATW.3.A2.FF.A2.

CMD.82.
DATW.FF.A2.FF.A2.FF.A2.

success!
```
**Additional info:**
 - MCU: [NUCLEO-G431KB](https://www.st.com/en/evaluation-tools/nucleo-g431kb.html)
 - [Arduino_Core_STM32](https://github.com/stm32duino/Arduino_Core_STM32)
 - Wireless module LAMBDA62 REV2 - [Datasheet](https://www.mouser.cn/datasheet/2/975/1551175771DS-LAMBDA62-2-1624682.pdf)
 - PlatformIO IDE version 2.3.0
 - Library version 4.3.0

Thank you for your time.

JoseCSNeto, 2021-03-17:
While I was trying to solve this issue, I decided to look into the library code to see if anything seemed suspicious.

That's when I tried changing the _startReceive_ function:
ORIGINAL
```c++
int16_t SX126x::startReceive(uint32_t timeout) {
  int16_t state = startReceiveCommon();
  RADIOLIB_ASSERT(state);

  // set RF switch (if present)
  _mod->setRfSwitchState(HIGH, LOW);

  // set mode to receive
  state = setRx(timeout);

  return(state);
}
```

MODIFIED BY ME
```c++
int16_t SX126x::startReceive(uint32_t timeout) {
  int16_t state;
    // set RF switch (if present)
  _mod->setRfSwitchState(HIGH, LOW);

  // set mode to receive
  state = setRx(timeout);
  state = startReceiveCommon();
  RADIOLIB_ASSERT(state);

  return(state);
}
```
and... it worked after that!

After this, I clearly noticed what the issue was. So, I tried solving the issue with the transmitter not working until I forced the _transmittedFlag_ to _true_. The solution is very similar to the mentioned above: move the _setRfSwitchState_ and _setTx_ to the start of the function.

Having solved this issue, I don't get why this problem only occurs on the STM32 Nucleo and not on the Arduinos I tested. Do you have any idea why?

jgromes, 2021-03-17:
I don't think this is the issue - in your modified code, you really shouldn't be setting the mode to RX via `setRX` *before* reception is properly configured using `startReceiveCommon`.

Did you try blocking transmit on the Nucleo? And if so, what was the result? Also, which Arduino are you using, Uno?

JoseCSNeto, 2021-03-18:
Blocking transmit on the Nucleo suffers from the same problem as the blocking receiver.

I used Arduino Uno and Nano during tests

jgromes, 2021-06-16:
I wasn't able to replicate any of the reported issues on Nucle L452RE-P, so I have to close for now.

closed 2021-06-16 by jgromes