jgromes/RadioLib #272: Example for using the new Arduino Portenta Vision Shield LoRa version

hpssjellis, 2021-03-17:
Any chance of connecting the new Arduino PortentaH7 Vision Shield LoRa version to RadioLib?

https://store.arduino.cc/usa/portenta-vision-shield-lora

jgromes, 2021-03-17:
In the words of a wise man: `¯\_(ツ)_/¯`

Arduino Portenta is a supported platform, meaning all RadioLib examples will compile on it. Whether it will actually work is a different question entirely, I don't have the Portenta board so I can't test it myself - though there was someone claiming RadioLib works fine on Portenta (#230).

Next is the LoRa module. Looking at the Vision shield schematics, it's CMWX1ZZABZ-093. This module combines an STM32L0 MCU and SX1276. If the SX1276 SPI bus is directly accessible from the outside, then RadioLib should be able to support it. Unfortunately from CMWX1ZZABZ-093 datasheet (https://wireless.murata.com/datasheet?/RFM/data/type_abz.pdf) it doesn't look like that's the case, and the SX1276 can only be accessed via the STM32.

Again, I don't have the module, and I can't test this, so until it can be proven otherwise, I don't think it will work. RadioLib needs that direct access to the SX1276.

closed 2021-03-17 by jgromes

hpssjellis, 2021-03-17:
Yes,Not sure why the pins arn't connected for LoRa use. I am looking into AT commands to do what I want.   Thanks for looking at it @jgromes