jgromes/RadioLib #273: CC1101 'Carrier Sense' or 'Tx on CCA'

Zmmfly, 2021-03-23:
**Is your feature request related to a problem? Please describe.**
Usually, we need transmit packet to many devices or receive.
But, when some devices transmit packet in same time, and no method to discern, it will cause interfere in channel, and transmit fail.

**Describe the solution you'd like**
Using 'Carrier Sense' or 'TX on CCA' to realize CSMA, it's will solution this problem.

**Additional context**
'AGCCTRL1' and 'AGCCTRL2' are importan Register for 'Carrier Sense' or 'TX on CCA'.

jgromes, 2021-03-23:
There's been a couple requests for carrier sense on different modules (e.g. https://github.com/jgromes/RadioLib/issues/68), but I'm not entirely sure this should be implemented in the library. I'm more inclined to think of RadioLib as collection of hardware drivers, in which case features like carrier sense, collision avoidance or delivery confirmation would be higher level of logic and should be implemented in higher ISO/OSI levels.

One issue is that different modules would have to implement carrier sense differently (e.g. all LoRa modules would have to implement CAD in LoRa mode, but RSSI measurement in FSK), so it would have to be added as some abstraction in `PhysicalLayer` and implemented in modules. Then there's the issue of collision resolution - I don't think it should be up to the library to decide what happens when there's a collision, or make sure it doesn't happen. Different wireless protocols might handle this differently, and again, RadioLib is mainly a collection of hardware drivers.

Overall, this is an issue of architecture - where's the boundary between RadioLib and user code. I remain sceptical about this, but if someone presents a good argument as to why it should be implemented here and not in higher layers, I will gladly change my mind.

jgromes, 2021-07-27:
In the end I've decided against this implementing any sort of carrier sense logic in RadioLib. The reason is that it's just too situation-dependent to create any sort of high-level interface. Besides, all the tools to achieve this are already present - almost every single module supported by the library has some a method that will return current RSSI. So if the application requires it, all the logic can be implemented quite easily on user side.

closed 2021-07-27 by jgromes