jgromes/RadioLib #276: Problem with 'receiveDirect()' on SX1276

gef30441, 2021-03-30:
I tried to read the incoming data continuously via the receiveDirect() function on the following board: [Heltec WiFi LoRa ESP32 V2](https://heltec.org/project/wifi-lora-32/).
The board utilizes the SX1276, which is wired like this to the ESP32:
- CS: GPIO18
- DIO0: GPIO26
- DIO1: GPI35
- DIO2: GPI35

I basically used the 'SX127x FSK Modem Example' and extracted the code i figured was enough to fit my needs.
You can see my code here:
[image]

Unfortunately when flashing the software on my board, I get the following error message:
[image]

Error code 16 translates to: [ERR_SPI_WRITE_FAILED](https://jgromes.github.io/RadioLib/group__status__codes.html#ga4e05af49324b9f065ae93f9664d51c2f).
The wiring can't be faulty (I hope at least) and I haven't found anything in the forums/RadioLib wiki that could help me.

I hope someone reading this is able to tell me what I did wrong here.

gef30441, 2021-03-30:
I forgot to mention that I already looked at the signals on DIO2 (DATA) and DIO1 (CLK), but nothing meaningful was transferred.

jgromes, 2021-03-30:
Double check the wiring, in your post it says that both DIO1 and DIO2 are connected to the same pin (GPI35), that seems wrong. Also, I suggest checking whether DIO2 is actually connected on that board, since on most LoRa/ESP32 boards, only DIO0 is always connected, and DIO1 is connected only sometimes. DIO2 is almost always unused.

Regarding your sketch, you're calling `receiveDirect` in the loop - you're basically spamming the radio with requests to set itself to Rx mode. Call `receiveDirect` once from setup, the radio will then remain in receive mode.

Finally, if neither of the above helps, please post debug mode output.

gef30441, 2021-03-31:
I double checked the wiring and realized I made a typo.
DIO2 is on GPI34 and DIO1 is on GPI35. 

Thanks for your response, I will try your suggestion asap and report if it worked or not.

gef30441, 2021-04-01:
So i just retried running the program with the following things changed:
- RadioLib_Debug active (RadioLib_Verbose inactive)
- receiveDirect() call in setup() instead of loop()

I get the following messages via Serial:
[image]

I also noticed something strange while trying around (RadioLib_Debug still active): 
_Code:_
[image]

_Output:_
[image]

Basically, I initialized the SX1276 with default settings and then started directReceive(). As you can see in the output, it didn't work (DCLK doesn't change periodically) - even though there was no error message.

jgromes, 2021-04-01:
The fact that it doesn't return the SPI error with default settings is interesting, I'll look into that.

You can't use Serial.prints to show the data and clock states, Serial takes way too long and the duration isn't always the same. You have to sample the data line based on an interrupt, and save them to a buffer, then print that. I'm currently working on implementing this sort of direct mode support.

gef30441, 2021-04-08:
I made another interesting discovery: I initialized the SX1276 with default settings, except for the frequency of 868.0 MHz. 
When calling the receiveDirect() once I get the error 16 (as stated above). But when calling it another time (directly after the first time in setup()) there's no error. I think register 0x1 still has its old contents saved to itself when checking the first time.

But calling receiveDirect() twice doesn't always work. For some specific parameters (in my case I changed the bitrate to 300kbps and the RxBandwidth to 3.1kHz via their specific function setBitRate() and setRxBandwidth()) error 16 remains.

gef30441, 2021-04-08:
I also tried printing the continuously received data.

[image]
[image]
[image]

Line 15 - 24: Naming the respective pins.
Line 28 - 43: Implementing the interrupt handler.
Line 47 - 48: I don't know if it's necessary to declare the pinModes explicitly here.
Line 51 : Attach interrupt to pin 35 at rising edge (at rising edge reading data is possible, at falling edge data is refreshed, more info in the [SX1276 datasheet](https://www.semtech.com/products/wireless-rf/lora-transceivers/sx1276).

How this basically works: In DCLK_interrupt() all contents in testvar are shifted a bit to the left. The current state of the data pin is saved to the "new" free bit on the right.
When the testvar is full (64 bits are filled in from data), the counter reached 64 and the receiveFull flag is set true. The counter gets a reset and the contents of testvar are deleted after getting copied to buffer.

The loop() constantly checks if the receiveFull flag is set. If yes, the buffer is printed via Serial.print and the receiveFull flag is reset.

jgromes, 2021-04-08:
> I initialized the SX1276 with default settings, except for the frequency of 868.0 MHz.
When calling the receiveDirect() once I get the error 16 (as stated above). But when calling it another time (directly after the first time in setup()) there's no error. I think register 0x1 still has its old contents saved to itself when checking the first time.

That's possible - I did notice already that some registers take time to update, especially the OpMode register. Though it seems strange it would depend on the value of the parameter.

Regarding the direct DIO sampling - that's very similar to what @yeckel has done with his direct mode D-Star receiver: https://github.com/yeckel/D-StarReceive

I'm also working on RadioLib implementation of this direct mode approach that would be available to all modules that can do this, not just SX127x.

yeckel, 2021-04-15:
Hi @gef30441 I've observed the same issue. When starting AFC I could not call ```setMode(SX127X_RX)``` however it seems to me that the RX mode **is actually set!** Just when the register is read back it contains the old bit value. Even after looong timeout. Probably you could just ignore that error :-D  Surprisingly ```setMode(SX127X_TX)``` works fine (changes the same bit).

jgromes, 2021-04-15:
I wasn't able to find anything in datasheet or errata that could explain this behavior, so let's just patch it in RadioLib - checking of that one bit was disabled in the latest commit, only for FSK RX.

@gef30441 @yeckel could you guys give it a try now?

yeckel, 2021-04-16:
Yes, it looks fine and works. I've tested it with my AFC. Good idea.

jgromes, 2021-04-17:
Thanks, looks like this has been resolved then - feel free to reopen if it hasn't.

closed 2021-04-17 by jgromes