jgromes/RadioLib #277: radio.receive(str) blocks forever with Arduino BLE Sense and CC1101

lasryaric, 2021-04-10:
**Describe the bug**
radio.receive(str) blocks for ever...
```
  int state = radio.receive(str);
```

**To Reproduce**
CC1101 receiver sample code on Arduino BLE sense 33

```
/*
   RadioLib CC1101 Receive Example

   This example receives packets using CC1101 FSK radio module.
   To successfully receive data, the following settings have to be the same
   on both transmitter and receiver:
    - carrier frequency
    - bit rate
    - frequency deviation
    - sync word

   For default module settings, see the wiki page
   https://github.com/jgromes/RadioLib/wiki/Default-configuration#cc1101

   For full API reference, see the GitHub Pages
   https://jgromes.github.io/RadioLib/
*/

//// include the library
#include <RadioLib.h>
//
//// CC1101 has the following connections:
//// CS pin:    10
//// GDO0 pin:  2
//// RST pin:   unused
//// GDO2 pin:  3 (optional)
CC1101 radio = new Module(10, 2, RADIOLIB_NC, 3);
//
//// or using RadioShield
//// https://github.com/jgromes/RadioShield
////CC1101 radio = RadioShield.ModuleA;

void setup() {
  Serial.begin(9600);
  while (!Serial) ;
  Serial.println("serial opened!!!");
  // initialize CC1101 with default settings
//  
  int state = radio.begin();
  if (state == ERR_NONE) {
    Serial.println("success! 2");
  } 
  else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
}

void loop() {
  
  Serial.print(F("[CC1101] Waiting for incoming transmission ... \n"));
  delay(1000);

  // you can receive data as an Arduino String
  String str;
  int state = radio.receive(str);
  
  Serial.print(F("[CC1101] Received!\n"));

  if (state == ERR_NONE) {
    // packet was successfully received
    Serial.println(F("success!"));

    // print the data of the packet
    Serial.print(F("[CC1101] Data:\t\t"));
    Serial.println(str);

    // print RSSI (Received Signal Strength Indicator)
    // of the last received packet
    Serial.print(F("[CC1101] RSSI:\t\t"));
    Serial.print(radio.getRSSI());
    Serial.println(F(" dBm"));

    // print LQI (Link Quality Indicator)
    // of the last received packet, lower is better
    Serial.print(F("[CC1101] LQI:\t\t"));
    Serial.println(radio.getLQI());

  } else if (state == ERR_CRC_MISMATCH) {
    // packet was received, but is malformed
    Serial.println(F("CRC error!"));

  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }

}
```

**Expected behavior**
`loop()` function should keep iterating and print the received data.

**Screenshots**
N/A

**Additional info (please complete):**
 - MCU: Arduino BLE sense 33 nano

 - Wireless module type CC1101 ( https://www.amazon.com/SMAKN%C2%AE-Wireless-Technical-External-Antenna/dp/B00U5TO37W )
 - Arduino IDE version  [e.g. 1.8.13]
 - Library version: radiolib 4.3.0

Any help in understanding why the function blocks forever would be really appreciated. I am an experienced software engineer but I have 0 knowledge in Arduino.

Thanks.

lasryaric, 2021-04-10:
I should add that it does fork for a few iteration, the `radio.receive(str)` function returns some data, but after a few iteration (~5) it starts blocking forever.

jgromes, 2021-04-10:
As far as I can tell, nothing in `CC1101::receive()` has the potential to be an infinite loop, all while cycles that wait for pin changes are guarded with timeouts. Unforunately I don't have a BLE Sense 33 Nano on hand to try it out. You could try running in [debug verbose](https://github.com/jgromes/RadioLib/wiki/Debug-mode), to see what's the last SPI exchange that is performed.

lasryaric, 2021-04-10:
Here is the output I get with the debug and verbose mode enabled.
The last line showing up in the logs is `14:54:33.680 -> [CC1101] Waiting for incoming transmission ... `, which is right before `radio.receive(str)`.

Any idea how about how to make progress from here?

Thanks

```
14:39:00.658 -> serial opened!!!
14:39:00.725 -> success! 2
14:39:00.725 -> [CC1101] Waiting for incoming transmission ... 
14:39:02.233 -> [CC1101] Received!
14:39:02.233 -> failed, code -6
14:39:02.233 -> [CC1101] Waiting for incoming transmission ... 
14:39:03.738 -> [CC1101] Received!
14:39:03.738 -> failed, code -6
14:39:03.738 -> [CC1101] Waiting for incoming transmission ... 
14:39:05.243 -> [CC1101] Received!
14:39:05.243 -> failed, code -6
14:39:05.243 -> [CC1101] Waiting for incoming transmission ... 
14:39:06.399 -> [CC1101] Received!
14:39:06.399 -> CRC error!
14:39:06.399 -> [CC1101] Waiting for incoming transmission ... 
14:39:07.923 -> [CC1101] Received!
14:39:07.923 -> failed, code -6
14:39:07.923 -> [CC1101] Waiting for incoming transmission ... 
14:39:09.348 -> [CC1101] Received!
14:39:09.348 -> CRC error!
14:39:09.348 -> [CC1101] Waiting for incoming transmission ... 
14:40:37.742 -> serial opened!!!
14:40:37.885 -> success! 2
14:40:37.885 -> [CC1101] Waiting for incoming transmission ... 
14:40:39.371 -> [CC1101] Received!
14:40:39.371 -> failed, code -6
14:40:39.371 -> [CC1101] Waiting for incoming transmission ... 
14:40:40.891 -> [CC1101] Received!
14:40:40.891 -> failed, code -6
14:40:40.891 -> [CC1101] Waiting for incoming transmission ... 
14:40:42.398 -> [CC1101] Received!
14:40:42.398 -> failed, code -6
14:40:42.398 -> [CC1101] Waiting for incoming transmission ... 
14:40:43.850 -> [CC1101] Received!
14:40:43.850 -> CRC error!
14:40:43.850 -> [CC1101] Waiting for incoming transmission ... 
14:40:45.354 -> [CC1101] Received!
14:40:45.354 -> failed, code -6
14:40:45.354 -> [CC1101] Waiting for incoming transmission ... 
14:40:46.853 -> [CC1101] Received!
14:40:46.853 -> failed, code -6
14:40:46.853 -> [CC1101] Waiting for incoming transmission ... 
14:48:00.955 -> serial opened!!!
14:48:01.093 -> success! 2
14:48:01.093 -> [CC1101] Waiting for incoming transmission ... 
14:48:02.609 -> [CC1101] Received!
14:48:02.609 -> failed, code -6
14:48:02.609 -> [CC1101] Waiting for incoming transmission ... 
14:48:04.093 -> [CC1101] Received!
14:48:04.093 -> failed, code -6
14:48:04.093 -> [CC1101] Waiting for incoming transmission ... 
14:48:05.593 -> [CC1101] Received!
14:48:05.593 -> failed, code -6
14:48:05.593 -> [CC1101] Waiting for incoming transmission ... 
14:48:07.106 -> [CC1101] Received!
14:48:07.106 -> failed, code -6
14:48:07.106 -> [CC1101] Waiting for incoming transmission ... 
14:48:08.611 -> [CC1101] Received!
14:48:08.611 -> failed, code -6
14:48:08.611 -> [CC1101] Waiting for incoming transmission ... 
14:48:09.816 -> [CC1101] Received!
14:48:09.816 -> CRC error!
[... 299 lines cut ...]
14:54:30.997 -> [CC1101] Received!
14:54:30.997 -> CRC error!
14:54:30.997 -> [CC1101] Waiting for incoming transmission ... 
14:54:32.014 -> R	2	6	
14:54:32.014 -> W	2	6	
14:54:32.014 -> R	2	6	
14:54:32.513 -> [CC1101] Received!
14:54:32.513 -> failed, code -6
14:54:32.513 -> [CC1101] Waiting for incoming transmission ... 
14:54:33.512 -> R	2	6	
14:54:33.512 -> W	2	6	
14:54:33.512 -> R	2	6	
14:54:33.646 -> R	7F	7F	
14:54:33.646 -> R	7	4	
14:54:33.646 -> R	7F	F6	31	1D	1	6D	51	3F	9C	35	0	EC	2	D1	81	59	F	65	80	49	44	58	7F	E3	C7	FC	A0	F0	7F	7F	59	63	11	5	4	FC	DF	F4	CF	F3	95	1	3	4	B7	9F	90	FC	7F	80	CE	86	7B	4F	19	E3	F9	35	87	28	A2	93	DC	39	D8	F6	31	1D	1	6D	51	3F	9C	35	0	EC	2	D1	81	59	F	65	80	49	44	58	7F	E3	C7	FC	A0	F0	7F	7F	59	63	11	5	4	FC	DF	F4	CF	F3	95	1	3	4	B7	9F	90	FC	7F	80	CE	86	7B	4F	19	E3	F9	35	87	28	A2	93	DC	39	
14:54:33.680 -> R	7F	39	
14:54:33.680 -> R	7F	D8	
14:54:33.680 -> [CC1101] Received!
14:54:33.680 -> success!
14:54:33.680 -> [CC1101] Data:		
14:54:33.680 -> [CC1101] RSSI:		-45.50 dBm
14:54:33.680 -> [CC1101] LQI:		88
14:54:33.680 -> [CC1101] Waiting for incoming transmission ... 

````

jgromes, 2021-04-14:
Looks like it gets stuck somewhere around here:

https://github.com/jgromes/RadioLib/blob/86ca714d0017419aebc5bae5b4e8010e71e66874/src/modules/CC1101/CC1101.cpp#L288-L297

Line 296 should print out 
```
14:54:33.512 -> R	2	6	
14:54:33.512 -> W	2	6	
14:54:33.512 -> R	2	6
```

when it gets executed. We don't see that, so taking into account the fact that Serial output has some delay, I'd say it's crashing somewhere around there. There's nothing but SPI transactions there, so the problem isn't obvious.

Digging around on Arduino forums a bit, I found the following: https://forum.arduino.cc/t/nano-33-ble-and-spi-stops-loop-running/683020/2

TL;DR - apparently, Nano 33 BLE will go to sleep mode in `delay()` which is quite a unique behavior and I have no idea what will it do with SPI. Could you try the Receive_Interrupt example to rule this out?

jgromes, 2021-05-02:
Closing as inactive - AFAIK this only happens on Nano 33 BLE, so I can't reproduce it. Feel free to reopen if there's more information.

closed 2021-05-02 by jgromes