jgromes/RadioLib #278: Memory management issue on AVR Arduino

ki4mcw, 2021-04-12:
Running the example RTTY program on an AVR Arduino can cause memory collisions. 

* Adafruit Feather 32u4 RFM95W at 433MHz (I believe the radio is an RFM96)
* Example program RTTY_Transmit.ino
* Changed "new Module" from SX1278 to RFM96
* Changed "rtty.begin" to use ITA2 mode, 2 stop bits (standard settings in fldigi)

With this combination, I get transmissions with lots of idling, and very few characters decoded (usually the first 2 of each line).

Using dynamic memory management, the _str value of an ITA2String is written successfully during construction, but is corrupted by the time the byteArray function is called. Changing the class definition of ITA2String in rtty.h to create _str with some value besides 1 (eg, 16 bytes) yields successful decodes, but crashes. Switching to static mode in BuildOpts.h instead compiles successfully, and yields 100% solid copy.

Maybe BuildOpts.h could specify static memory mode for AVR platforms? I have some 328s I could experiment with, but unfortunately no other radios except for the ones built onto the Feathers.

Thanks!

ki4mcw, 2021-04-13:
I should mention, the main issue - incomplete transmissions using an ATMega32u4 + RTTY + ITA2 - is not immediately obvious. The radio keys up, and it does send RTTY out, but translates the overwritten data (3rd char and beyond in a print or println) as non-compliant for ITA2, so it sends nulls instead of those characters. Those nulls fill in the time that would normally be spent sending characters, so it looks like a receive/decode problem. Debugging in the byteArray function exposed the issue.

If you use ASCII mode instead of ITA2, this error might never appear. Likewise if you use a device with more memory, or more robust dynamic memory allocation (it may even be a gcc bug, not sure). 

Thanks!

jgromes, 2021-04-13:
Looking over ITA2String implementation, it's clear why this happens - `_str` never gets re-allocated to the correct size. The entire `char* _str = new char[1]` is pretty dumb, I have no idea why I did that.

jgromes, 2021-04-14:
@ki4mcw this should be fixed now, unfortunately I don't have a Feather to test with, could you give it a shot?

ki4mcw, 2021-04-15:
Fixed! I added "toupper()" on lines 47 and 63 of RTTY.cpp, and the example code ran great (without those, I had to edit all of the strings manually, or the nulling of the lower-case letters looked sort of like the original issue).

In case it might be useful - my Feathers with in-built RFM96s require this line to specify pins:

     RFM96 radio = new Module(8, 7, 4);

Thank you, Jan!

jgromes, 2021-04-15:
Glad it worked, thanks for reporting it!

closed 2021-04-15 by jgromes