jgromes/RadioLib #280: More robust AFC

yeckel, 2021-04-13:
Hi Honzo, do you think there is a way to implement a more robust AFC? My SX1278 board has offset 2.5kHz! When I transmit from ICOM radio, then I have to be within +/- 500Hz to be able to receive it (preamble and sync word). The icom is fine with +/- 6kHz. Today I was hunting for a weather balloon with [rdz_ttgo_sonde](https://github.com/dl9rdz/rdz_ttgo_sonde) and the same SX1278. The sonde decoder was reporting frequency offset from the RS41 some 2.5kHz and receiving just fine.

dl9rdz is using separate values for  SX127X_REG_AFC_BW(12.5kHz) and SX127X_REG_RX_BW(6.3kHz) which suits D-Star as well. He also setts SX127X_REG_RX_CONFIG to ```SX127X_AGC_AUTO_ON | SX127X_AFC_AUTO_ON | SX127X_RX_TRIGGER_PREAMBLE_DETECT```. However when I do that I'm getting -16 (SPI write error) by setMode(SX127X_RX) in SX127x:: receiveDirect() even with several delays all around :-D

Any ideas? I'm also not really getting what's the difference between AFC and FEI, sounds the same to me. In the RadioLib you are using just FEI, but the output of the getFrequencyError() is basically just a random generator for me.

Thanks!

My whole code [D-StarBeacon](https://github.com/yeckel/D-StarBeacon)

yeckel, 2021-04-14:
I was able to test a modified dl9rdz code for RS41 and I'm getting preambled&synced when ICOM transmits from 438.784MHz to 438.809MHz, which gives **25kHz** with  SX127X_REG_AFC_BW(12.5kHz) and SX127X_REG_RX_BW(6.3kHz). Which is awesome. Well my TX and RX sit next to each other. Now I'll try comparing the registers between RadioLib and rdz_ttgo_sonde and see if I can make a patch from it.

jgromes, 2021-04-14:
This -16 error after setting mode to Rx in FSK keeps popping up lately.

Both AFC and FEI are feautures of the SX127x, and aren't very well explained in the datasheets, so it is possible they're not correctly configured. From what I gathered, I understood AFC attempts to automatically correct the frequency during transmission, while FEI reports the received carrier frequency offset from the expected frequency.

> Now I'll try comparing the registers between RadioLib and rdz_ttgo_sonde and see if I can make a patch from it

Thanks! Please let me know what you find, or feel free to open a PR.

yeckel, 2021-04-15:
Well, I've analyzed register dumps from RadioLib and from DL1RDZ and found the key difference. RDZ is setting AFC on :-D and not checking writes into ```SX127X_REG_OP_MODE```.

When the op mode is set to 0b101 (RX) it stays 0b100 *forever*. I would say it's an HW or datasheet bug. You have a [PR](https://github.com/jgromes/RadioLib/pull/282) but I'm afraid some API change would be necessary.

jgromes, 2021-04-17:
Implemented in #282 - many thanks @yeckel!

closed 2021-04-17 by jgromes