jgromes/RadioLib #281: Error code -16

rutwikganagi, 2021-04-14:
When I execute radio.transmit() function after using radio.beginFSK(), I get "[SX1278] Failed to receive the packet, code -16" , however when i change the function to radio.begin() there is no error. Lora modulation is taking place but FSK modulation is not taking place. "[SX1278]  Initializing... Success!" is also displayed. Can you please solve this issue

jgromes, 2021-04-14:
> Can you please solve this issue

Unfortunately no, unless you provide more information. Use the "Module not working" issue template and fill in the required information, such as debug mode output, microcontroller type etc.

1. **Read [CONTRIBUTING.md](https://github.com/jgromes/RadioLib/blob/master/CONTRIBUTING.md)!** Issues that do not follow this document will be closed/locked/deleted/ignored.
2. RadioLib has a [Wiki](https://github.com/jgromes/RadioLib/wiki) and an extensive [API documentation](https://jgromes.github.io/RadioLib/). You might find a solution to your issue there.
3. Make sure you're using the latest release of the library! Releases can be found [here](https://github.com/jgromes/RadioLib/releases).
4. Use [Arduino forums](https://forum.arduino.cc/) to ask generic questions about wireless modules, wiring, usage, etc. Only create issues for problems specific to RadioLib!
5. Error codes, their meaning and how to fix them can be found on [this page](https://jgromes.github.io/RadioLib/group__status__codes.html).

**Sketch that is causing the module fail**

```c++
paste the sketch here, even if it is an unmodified example code
```

**Hardware setup**
Wiring diagram, schematic, pictures etc.

**Debug mode output**
Enable all [debug levels](https://github.com/jgromes/RadioLib/wiki/Debug-mode) and paste the Serial monitor output here.

**Additional info (please complete):**
 - MCU: [e.g. Arduino Uno, ESP8266 etc.]
 - Link to Arduino core: [e.g. https://github.com/stm32duino/Arduino_Core_STM32 when using official STM32 core. See readme for links to all supported cores]
 - Wireless module type [e.g. CC1101, SX1268, etc.]
 - Arduino IDE version  [e.g. 1.8.5]
 - Library version [e.g. 3.0.0]

rutwikganagi, 2021-04-15:
**Describe the error:**

radio.transmit() is not working for FSK mode, but works for LoRa mode
```
int state = radio.transmit("Hello World!");
```
Error code -16 is shown on the terminal window

**Entire code:**
```
// include the library
#include <RadioLib.h>

// SX1278 has the following connections:
// NSS pin:   10
// DIO0 pin:  2
// RESET pin: 9
// DIO1 pin:  3
SX1278 radio = new Module(10, 2, 9, 3);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1278 fsk = RadioShield.ModuleA;

void setup() {
  Serial.begin(9600);

  // initialize SX1278 FSK modem with default settings
  Serial.print(F("[SX1278] Initializing ... "));
  int state = radio.beginFSK();
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  // if needed, you can switch between LoRa and FSK modes
  //
  // radio.begin()       start LoRa mode (and disable FSK)
  // radio.beginFSK()    start FSK mode (and disable LoRa)

  // the following settings can also
  // be modified at run-time
  state = radio.setFrequency(433.5);
  state = radio.setBitRate(100.0);
  state = radio.setFrequencyDeviation(10.0);
  state = radio.setRxBandwidth(250.0);
  state = radio.setOutputPower(10.0);
  state = radio.setCurrentLimit(100);
  state = radio.setDataShaping(RADIOLIB_SHAPING_0_5);
  uint8_t syncWord[] = {0x01, 0x23, 0x45, 0x67,
                        0x89, 0xAB, 0xCD, 0xEF};
  state = radio.setSyncWord(syncWord, 8);
  if (state != ERR_NONE) {
    Serial.print(F("Unable to set configuration, code "));
    Serial.println(state);
    while (true);
  }

  // FSK modulation can be changed to OOK
  // NOTE: When using OOK, the maximum bit rate is only 32.768 kbps!
  //       Also, data shaping changes from Gaussian filter to
  //       simple filter with cutoff frequency. Make sure to call
  //       setDataShapingOOK() to set the correct shaping!
 /* state = radio.setOOK(true);
  state = radio.setDataShapingOOK(1);
  if (state != ERR_NONE) {
    Serial.print(F("Unable to change modulation, code "));
    Serial.println(state);
    while (true);
  }*/

  #warning "This sketch is just an API guide! Read the note at line 6."
}

void loop() {
  // FSK modem can use the same transmit/receive methods
  // as the LoRa modem, even their interrupt-driven versions
  // NOTE: FSK modem maximum packet length is 63 bytes!

  // transmit FSK packet
  int state = radio.transmit("Hello World!");
  /*
    byte byteArr[] = {0x01, 0x23, 0x45, 0x67,
                      0x89, 0xAB, 0xCD, 0xEF};
    int state = radio.transmit(byteArr, 8);
  */
  if (state == ERR_NONE) {
    Serial.println(F("[SX1278] Packet transmitted successfully!"));
  } else if (state == ERR_PACKET_TOO_LONG) {
    Serial.println(F("[SX1278] Packet too long!"));
  } else if (state == ERR_TX_TIMEOUT) {
    Serial.println(F("[SX1278] Timed out while transmitting!"));
  } else {
    Serial.println(F("[SX1278] Failed to transmit packet, code "));
    Serial.println(state);
  }
}
```
**Output:**
[image]

**Additional Info:**
MCU: Arduino UNO
Wireless Module Type: LoRa Ra-02 SX1278
Arduino IDE Version: 1.8.13
RadioLib Version: 4.3.0

jgromes, 2021-04-16:
Please post output with debug mode enabled, as described here: https://github.com/jgromes/RadioLib/wiki/Debug-mode

rutwikganagi, 2021-04-17:
**Debug Mode:**

[image]

**Hardware setup**

[image]

**Connections:**
NSS-10
MOSI-11
MISO-12
SCK-13

jgromes, 2021-04-17:
It looks like your SX1278 is unable to enter transmit mode. Couple of notes on your setup:

1. You have a 5V device (Arduino Uno) connected directly to a device with 3.3V logic (SX1278). Sooner or later, that will damage the LoRa module. You have to use a logic level shifter to convert the 5V signals from Arduino to 3.3V signals for the SX1278.

2. Never run a radio without an antenna. If you do, and you transmit, you will damage or destroy the power amplifier (which could be why your SX1278 cannot enter transmit mode).

3. What's the power source for the SX1278? It's not shown in the picture.

rutwikganagi, 2021-04-17:
I connected 3.3V of Arduino Uno pin to 3.3V pin of sx1278

I used laptop as power source connected using Arduino uno cable
I didn't use any other external power source

I'll try once with antenna and see if it works

Thank you for your help!!

jgromes, 2021-05-02:
Closing as inactive, feel free to reopen later.

closed 2021-05-02 by jgromes

rutwikganagi, 2021-06-28:
I am getting the same error even after using an antenna.
3.3V pin of Arduino uno is connected to 3.3V pin of Ra-02 SX1278.

**Hardware Setup:**

[image]

**Output:**

[image]

**Debug Mode:**

[image]

jgromes, 2021-06-28:
As per my previous comment:

> You have a 5V device (Arduino Uno) connected directly to a device with 3.3V logic (SX1278). Sooner or later, that will damage the LoRa module. You have to use a logic level shifter to convert the 5V signals from Arduino to 3.3V signals for the SX1278.

I think you have damaged your radio module, either by attempting to transmit without an antenna, or by connecting it directly to a 5V Arduino. This seems to be supported by the debug output, as the bit that's supposed to set the device to Tx mode is stuck at 0.

rutwikganagi, 2021-06-28:
If i had damaged the radio module, LoRa Modulation should not be working, but it's working.

**Code:**

```*
   RadioLib SX127x Transmit Example

   This example transmits packets using SX1278 LoRa radio module.
   Each packet contains up to 256 bytes of data, in the form of:
    - Arduino String
    - null-terminated char array (C-string)
    - arbitrary binary data (byte array)

   Other modules from SX127x/RFM9x family can also be used.

   For default module settings, see the wiki page
   https://github.com/jgromes/RadioLib/wiki/Default-configuration#sx127xrfm9x---lora-modem

   For full API reference, see the GitHub Pages
   https://jgromes.github.io/RadioLib/
*/

// include the library
#include <RadioLib.h>

// SX1278 has the following connections:
// NSS pin:   10
// DIO0 pin:  2
// RESET pin: 9
// DIO1 pin:  3
SX1278 radio = new Module(10, 2, 9, 3);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1278 radio = RadioShield.ModuleA;

void setup() {
  Serial.begin(9600);

  // initialize SX1278 with default settings
  Serial.print(F("[SX1278] Initializing ... "));
  int state = radio.begin();
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  // some modules have an external RF switch
  // controlled via two pins (RX enable, TX enable)
  // to enable automatic control of the switch,
  // call the following method
  // RX enable:   4
  // TX enable:   5
  /*
    radio.setRfSwitchPins(4, 5);
  */
}

void loop() {
  Serial.print(F("[SX1278] Transmitting packet ... "));

  // you can transmit C-string or Arduino string up to
  // 256 characters long
  // NOTE: transmit() is a blocking method!
  //       See example SX127x_Transmit_Interrupt for details
  //       on non-blocking transmission method.
  int state = radio.transmit("Hello World!");

  // you can also transmit byte array up to 256 bytes long
  /*
    byte byteArr[] = {0x01, 0x23, 0x45, 0x67, 0x89, 0xAB, 0xCD, 0xEF};
    int state = radio.transmit(byteArr, 8);
  */

  if (state == ERR_NONE) {
    // the packet was successfully transmitted
    Serial.println(F(" success!"));

    // print measured data rate
    Serial.print(F("[SX1278] Datarate:\t"));
    Serial.print(radio.getDataRate());
    Serial.println(F(" bps"));

  } else if (state == ERR_PACKET_TOO_LONG) {
    // the supplied packet was longer than 256 bytes
    Serial.println(F("too long!"));

  } else if (state == ERR_TX_TIMEOUT) {
    // timeout occurred while transmitting packet
    Serial.println(F("timeout!"));

  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }

  // wait for a second before transmitting again
  delay(1000);
}
```

**Output:(LoRa Mode)**

[image]

**This is the antenna which i am using:**

[image]

jgromes, 2021-06-28:
> If i had damaged the radio module, LoRa Modulation should not be working, but it's working.

Not neccessarily, only the FSK modem could be damaged. Regardless, I wouldn't trust this module to be 100% working anymore, and I would definitely add proper conversion from 5V logic to 3.3V logic next time (which you still lack).

The antenna might be sold as 433 MHz, but it definitely looks incorrectly labeled. Better than nothing I guess.

rutwikganagi, 2021-09-11:
After i bought new Ra-02 SX1278 modules and Bi-Directional logic level converter, the Transmitter Code is working absolutely fine, but for the receiver code, it is showing Timed out while waiting for the packet!

**Hardware setup:**

[image]

**Transmitter Code:**
```
/*
   RadioLib SX127x FSK Modem Example

   This example shows how to use FSK modem in SX127x chips.

   NOTE: The sketch below is just a guide on how to use
         FSK modem, so this code should not be run directly!
         Instead, modify the other examples to use FSK
         modem and use the appropriate configuration
         methods.

   For default module settings, see the wiki page
   https://github.com/jgromes/RadioLib/wiki/Default-configuration#sx127xrfm9x---fsk-modem

   For full API reference, see the GitHub Pages
   https://jgromes.github.io/RadioLib/
*/

// include the library
#include <RadioLib.h>
#include <string.h>
// SX1278 has the following connections:
// NSS pin:   10
// DIO0 pin:  2
// RESET pin: 9
// DIO1 pin:  3
SX1278 radio = new Module(10, 2, 9, 3);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1278 fsk = RadioShield.ModuleA;

void setup() {
  Serial.begin(9600);

  // initialize SX1278 FSK modem with default settings
  Serial.print(F("[SX1278] Initializing ... "));
  int state = radio.beginFSK();
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  // if needed, you can switch between LoRa and FSK modes
  //
  // radio.begin()       start LoRa mode (and disable FSK)
  // radio.beginFSK()    start FSK mode (and disable LoRa)

  // the following settings can also
  // be modified at run-time
  state = radio.setFrequency(434.0);
  state = radio.setBitRate(100.0);
  state = radio.setFrequencyDeviation(10.0);
  state = radio.setRxBandwidth(250.0);
  state = radio.setOutputPower(10.0);
  state = radio.setCurrentLimit(100);
  state = radio.setDataShaping(RADIOLIB_SHAPING_0_5);
  uint8_t syncWord[] = {0x01, 0x23, 0x45, 0x67,
                        0x89, 0xAB, 0xCD, 0xEF};
[... 147 lines cut ...]
    // of the last received packet
    Serial.print(F("[SX1278] SNR:\t\t\t"));
    Serial.print(radio.getSNR());
    Serial.println(F(" dB"));

    // print frequency error
    // of the last received packet
    Serial.print(F("[SX1278] Frequency error:\t"));
    Serial.print(radio.getFrequencyError());
    Serial.println(F(" Hz"));

  } else if (state == ERR_RX_TIMEOUT) {
    Serial.println(F("[SX1278] Timed out while waiting for packet!"));
  } else {
    Serial.println(F("[SX1278] Failed to receive packet, code "));
    Serial.println(state);
  }

 
} 
```

**Receiver Output:**

[image]

jgromes, 2021-09-12:
Timeout simply menas the received got no packet within some time period (5x 256 B at the configured bitrate). Since you seem to be using a quite high bitrate (100 kbps), the packets and the timeout period are very short. It's a lot better to use interrupt-driven reception, since that doesn't use the blocking `receive` method.

rutwikganagi, 2021-09-13:
**Thanks A lot for you help!!**

I am able to correctly receive the packets with the interrupt codes

I had one more doubt
I am not able to get SNR and Frequency error for the received packets using FSK modulation

Is it because there is no built in registers for SNR and Frequency error using FSK modulation?

**Receiver Output:**
[image]

jgromes, 2021-09-13:
> Is it because there is no built in registers for SNR and Frequency error using FSK modulation?

That's correct, SNR and frequency error are only implemented in the chip for LoRa modulation.

rutwikganagi, 2021-09-14:
Is there any way i can measure data rate of transmitting packets for the interrupt code?
It is showing 0 using getDataRate() function

Transmitter Code:
```/*
   RadioLib SX127x Transmit with Interrupts Example

   This example transmits LoRa packets with one second delays
   between them. Each packet contains up to 256 bytes
   of data, in the form of:
    - Arduino String
    - null-terminated char array (C-string)
    - arbitrary binary data (byte array)

   Other modules from SX127x/RFM9x family can also be used.

   For default module settings, see the wiki page
   https://github.com/jgromes/RadioLib/wiki/Default-configuration#sx127xrfm9x---lora-modem

   For full API reference, see the GitHub Pages
   https://jgromes.github.io/RadioLib/
*/

// include the library
#include <RadioLib.h>

// SX1278 has the following connections:
// NSS pin:   10
// DIO0 pin:  2
// RESET pin: 9
// DIO1 pin:  3
SX1278 radio = new Module(10, 2, 9, 3);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1278 radio = RadioShield.ModuleA;

// save transmission state between loops
int transmissionState = ERR_NONE;
int count=1;
String s = String(count);
void setup() {
  Serial.begin(9600);

  // initialize SX1278 with default settings
  Serial.print(F("[SX1278] Initializing ... "));
  int state = radio.beginFSK();
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  // set the function that will be called
  // when packet transmission is finished
  radio.setDio0Action(setFlag);

  // start transmitting the first packet
  Serial.print(F("[SX1278] Sending  packet No: "));
  Serial.println(s);
  
  // you can transmit C-string or Arduino string up to
  // 256 characters long
  
  transmissionState = radio.startTransmit("Hello World:"+s);
  // print measured data rate
  Serial.print(F("[SX1278] Datarate:\t"));
  Serial.print(radio.getDataRate());
  Serial.println(F(" bps"));

  // you can also transmit byte array up to 256 bytes long
  /*
    byte byteArr[] = {0x01, 0x23, 0x45, 0x67,
                      0x89, 0xAB, 0xCD, 0xEF};
    state = radio.startTransmit(byteArr, 8);
  */
}

// flag to indicate that a packet was sent
volatile bool transmittedFlag = false;

// disable interrupt when it's not needed
volatile bool enableInterrupt = true;

// this function is called when a complete packet
// is transmitted by the module
// IMPORTANT: this function MUST be 'void' type
//            and MUST NOT have any arguments!
void setFlag(void) {
  // check if the interrupt is enabled
  if(!enableInterrupt) {
    return;
  }

  // we sent a packet, set the flag
  transmittedFlag = true;
}

int count2=1;

void loop() {
  // check if the previous transmission finished
  if(transmittedFlag) {
    // disable the interrupt service routine while
    // processing the data
    enableInterrupt = false;

    // reset flag
    transmittedFlag = false;

    if (transmissionState == ERR_NONE) {
      // packet was successfully sent
      Serial.println(F("transmission finished!"));

      // NOTE: when using interrupt-driven transmit method,
      //       it is not possible to automatically measure
      //       transmission data rate using getDataRate()

    } else {
      Serial.print(F("failed, code "));
      Serial.println(transmissionState);

    }

    // NOTE: in FSK mode, SX127x will not automatically
    //       turn transmitter off after sending a packet
    //       set mode to standby to ensure we don't jam others
    //radio.standby()

    // wait a second before transmitting again
    delay(1000);

    count2=count2+1;
    String s2 = String(count2);
    
    // send another one
    Serial.print(F("[SX1278] Sending  packet No: "));
    Serial.println(s2);

    // you can transmit C-string or Arduino string up to
    // 256 characters long
    transmissionState = radio.startTransmit("Hello World : "+s2);

    // print measured data rate
    Serial.print(F("[SX1278] Datarate:\t"));
    Serial.print(radio.getDataRate());
    Serial.println(F(" bps"));

    // you can also transmit byte array up to 256 bytes long
    /*
      byte byteArr[] = {0x01, 0x23, 0x45, 0x67,
                        0x89, 0xAB, 0xCD, 0xEF};
      int state = radio.startTransmit(byteArr, 8);
    */

    // we're ready to send more packets,
    // enable interrupt service routine
    enableInterrupt = true;
    
  }
}```

jgromes, 2021-09-14:
You can't use `getDataRate()` wiht interrupts, since it relies on the entire transmission process to be handled by the library, so if you need to know the effective data rate, you will have to implement it yourself - save one timestamp of when packet transmission starts, another one when it finishes, then subtract the two and divide packet length with that number.

rutwikganagi, 2021-09-16:
I don't know how to do it

Can you please modify the below code and send it to me
It would be really helpful

**Thank You in advance**!

**Transmitter Interrupt Code:**

```
/*
   RadioLib SX127x Transmit with Interrupts Example

   This example transmits LoRa packets with one second delays
   between them. Each packet contains up to 256 bytes
   of data, in the form of:
    - Arduino String
    - null-terminated char array (C-string)
    - arbitrary binary data (byte array)

   Other modules from SX127x/RFM9x family can also be used.

   For default module settings, see the wiki page
   https://github.com/jgromes/RadioLib/wiki/Default-configuration#sx127xrfm9x---lora-modem

   For full API reference, see the GitHub Pages
   https://jgromes.github.io/RadioLib/
*/

// include the library
#include <RadioLib.h>

// SX1278 has the following connections:
// NSS pin:   10
// DIO0 pin:  2
// RESET pin: 9
// DIO1 pin:  3
SX1278 radio = new Module(10, 2, 9, 3);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1278 radio = RadioShield.ModuleA;

// save transmission state between loops
int transmissionState = ERR_NONE;
int count=1;
String s = String(count);
void setup() {
  Serial.begin(9600);

  // initialize SX1278 with default settings
  Serial.print(F("[SX1278] Initializing ... "));
  int state = radio.beginFSK();
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  // set the function that will be called
  // when packet transmission is finished
  radio.setDio0Action(setFlag);

  // start transmitting the first packet
  Serial.print(F("[SX1278] Sending  packet No: "));
  Serial.println(s);
  
  // you can transmit C-string or Arduino string up to
  // 256 characters long
  
  transmissionState = radio.startTransmit("Hello World:"+s);
  // print measured data rate
  Serial.print(F("[SX1278] Datarate:\t"));
  Serial.print(radio.getDataRate());
  Serial.println(F(" bps"));

  // you can also transmit byte array up to 256 bytes long
  /*
    byte byteArr[] = {0x01, 0x23, 0x45, 0x67,
                      0x89, 0xAB, 0xCD, 0xEF};
    state = radio.startTransmit(byteArr, 8);
  */
}

// flag to indicate that a packet was sent
volatile bool transmittedFlag = false;

// disable interrupt when it's not needed
volatile bool enableInterrupt = true;

// this function is called when a complete packet
// is transmitted by the module
// IMPORTANT: this function MUST be 'void' type
//            and MUST NOT have any arguments!
void setFlag(void) {
  // check if the interrupt is enabled
  if(!enableInterrupt) {
    return;
  }

  // we sent a packet, set the flag
  transmittedFlag = true;
}

int count2=1;

void loop() {
  // check if the previous transmission finished
  if(transmittedFlag) {
    // disable the interrupt service routine while
    // processing the data
    enableInterrupt = false;

    // reset flag
    transmittedFlag = false;

    if (transmissionState == ERR_NONE) {
      // packet was successfully sent
      Serial.println(F("transmission finished!"));

      // NOTE: when using interrupt-driven transmit method,
      //       it is not possible to automatically measure
      //       transmission data rate using getDataRate()

    } else {
      Serial.print(F("failed, code "));
      Serial.println(transmissionState);

    }

    // NOTE: in FSK mode, SX127x will not automatically
    //       turn transmitter off after sending a packet
    //       set mode to standby to ensure we don't jam others
    //radio.standby()

    // wait a second before transmitting again
    delay(1000);

    count2=count2+1;
    String s2 = String(count2);
    
    // send another one
    Serial.print(F("[SX1278] Sending  packet No: "));
    Serial.println(s2);

    // you can transmit C-string or Arduino string up to
    // 256 characters long
    transmissionState = radio.startTransmit("Hello World : "+s2);

    // print measured data rate
    Serial.print(F("[SX1278] Datarate:\t"));
    Serial.print(radio.getDataRate());
    Serial.println(F(" bps"));

    // you can also transmit byte array up to 256 bytes long
    /*
      byte byteArr[] = {0x01, 0x23, 0x45, 0x67,
                        0x89, 0xAB, 0xCD, 0xEF};
      int state = radio.startTransmit(byteArr, 8);
    */

    // we're ready to send more packets,
    // enable interrupt service routine
    enableInterrupt = true;
    
  }
}

rutwikganagi, 2021-09-21:
Is there any way I can measure SNR for FSK modulation since there is no built in registers?

jgromes, 2021-09-21:
> Is there any way I can measure SNR for FSK modulation since there is no built in registers?

Possibly, using an external signal analyzer or SDR. Not with the SX1278 though.