jgromes/RadioLib #292: playing with registers

artouste, 2021-04-25:
Hello
I often use your Radiolib lib.
Currently I am experimenting with semtech sx1278 modules and am having trouble playing with registers.

I am currently trying to finely configure FSK / OOK mode and find that I cannot (I do not know) use with the sx1278 module the following functions:
SPIgetRegValue ()
SPIreadRegister ()
SPIwriteRegister ()
I get this error in compilation
'class SX1278' has no member named 'SPIreadRegister'
Can you tell me the correct method?
Thank you
Phil Artouste  
artoustephil@orange.fr

jgromes, 2021-04-25:
You haven't provided any code that leads to this, so I have to assume you're doing something like:

```c++
radio.SPIgetRegValue();
```

You will have to enable `RADIOLIB_LOW_LEVEL` in BuildOpt.h to access the low-level SPI methods from your high-level sketch. You really shouldn't do this, unless you know what you're doing.

When that macro is enabled, you can call low-level SPI access methods like this:

```c++
radio._mod->SPIgetRegValue();
```

artouste, 2021-04-25:
Thank you for your quick reply
It is impeccable!
  Normally I know more or less what I'm doing and if it smokes I have a big red button on my hand : D
Artouste

closed 2021-04-25 by jgromes