jgromes/RadioLib #296: RFM69 loosing last byte of payload

pilotak, 2021-05-02:
**Describe the bug**
When sending 8 bytes over RFM69 only 7 is received, it looks like the first received byte is payload length but then it should be payload but payload length is repeated instead followed by payload but the last byte is trimmed.

> Maybe due to this offset error AES is not working - nothing is received (i can see something is transmitted on SDR). Once this is sorted i will submit another issue if the problem persists.

**To Reproduce**
# Transmitter
```cpp
#include <RadioLib.h>

RF69 radio = new Module(10, 6, 5);

void setup() {
  Serial.begin(9600);

  while (!Serial) {
    ;
  }

  Serial.print(F("[RF69] Initializing ... "));
  int state = radio.begin(868.0, 48.0, 50.0, 125.0, 5, 16);
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
}

void loop() {
  Serial.print(F("[RF69] Transmitting packet ... "));

  byte byteArr[] = {0x0A, 0x23, 0x45, 0x67, 0x89, 0xAB, 0xCD, 0xEF};
  int state = radio.transmit(byteArr, 8);

  if (state == ERR_NONE) {
    Serial.println(F("success!"));

  } else if (state == ERR_PACKET_TOO_LONG) {
    Serial.println(F("too long!"));

  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }

  delay(1000);
}
```

## LOG
```
[RF69] Initializing ... R	10	24	
M	RF69
R	1	84	
W	1	84	
R	1	84	
R	1	84	
W	1	84	
R	1	84	
R	13	1A	
W	13	1A	
R	13	1A	
R	2	0	
W	2	0	
R	2	0	
R	2	0	
W	2	3	
R	2	3	
R	29	E4	
W	29	E4	
R	29	E4	
W	28	10	
R	26	7	
W	26	7	
R	26	7	
R	37	90	
W	37	90	
R	37	90	
R	3D	3	
W	3D	3	
R	3D	3	
R	3D	3	
W	3D	2	
R	3D	2	
R	38	40	
W	38	FF	
R	38	FF	
R	3C	8F	
W	3C	8F	
R	3C	8F	
R	2A	FF	
W	2A	FF	
R	2A	FF	
R	2B	FF	
W	2B	FF	
R	2B	FF	
R	6F	30	
W	6F	30	
R	6F	30	
R	1	84	
W	1	84	
R	1	84	
W	7	D9	
W	8	0	
W	9	0	
R	1	84	
W	1	84	
R	1	84	
R	3	2	
W	3	2	
R	3	2	
R	4	9A	
W	4	9A	
R	4	9A	
R	1	84	
W	1	84	
R	1	84	
R	19	42	
W	19	42	
R	19	42	
R	1	84	
W	1	84	
R	1	84	
R	5	3	
W	5	3	
R	5	3	
R	6	33	
W	6	33	
R	6	33	
R	1	84	
W	1	84	
R	1	84	
R	11	97	
W	11	97	
R	11	97	
W	2C	0	
R	2D	2	
W	2D	2	
R	2D	2	
R	37	90	
W	37	90	
R	37	90	
R	38	FF	
W	38	40	
R	38	40	
R	2E	88	
W	2E	88	
R	2E	88	
W	2F	12	AD	
R	1	84	
W	1	84	
R	1	84	
R	2	3	
W	2	0	
R	2	0	
R	1	84	
W	1	84	
R	1	84	
R	37	90	
W	37	90	
R	37	90	
R	37	90	
W	37	90	
R	37	90	
success!
W	3E	0	1	2	3	4	5	6	7	8	9	A	B	C	D	E	F	
R	3D	2	
W	3D	3	
R	3D	3	
[RF69] Transmitting packet ... R	1	84	
W	1	84	
R	1	84	
R	25	0	
W	25	0	
R	25	0	
W	27	FF	
W	28	FF	
W	0	8	
W	0	8	
R	37	90	
W	0	A	23	45	67	89	AB	CD	EF	
R	1	84	
W	1	8C	
R	1	8C	
R	1	8C	
W	1	84	
R	1	84	
W	27	FF	
W	28	FF	
success!
```

# Receiver
```cpp
#include <RadioLib.h>

RF69 radio = new Module(53, 2, 7);

void setup() {
  Serial.begin(9600);

  Serial.print(F("[RF69] Initializing ... "));
  int state = radio.begin(868.0, 48.0, 50.0, 125.0, 1, 16);

  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
}

void loop() {
  Serial.print(F("[RF69] Waiting for incoming transmission ... "));

    byte byteArr[32];
    memset(byteArr, 0, 32);
    int state = radio.receive(byteArr, 32);
  

  if (state == ERR_NONE) {
    Serial.println(F("success!"));

    Serial.print(F("[RF69] Data:"));
    Serial.print(state);
    Serial.print("\t\t");

    for(uint8_t i = 0; i < 32;i++){
      Serial.print(byteArr[i], HEX);
      Serial.print(' ');
    }
    Serial.println();

  } else if (state == ERR_RX_TIMEOUT) {
    Serial.println(F("timeout!"));

  } else if (state == ERR_CRC_MISMATCH) {
    Serial.println(F("CRC error!"));

  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);

  }
}
```
## LOG
```
[RF69] Initializing ... R	10	24	
M	RF69
R	1	90	
W	1	84	
R	1	84	
R	1	84	
W	1	84	
R	1	84	
R	13	1A	
W	13	1A	
R	13	1A	
R	2	0	
W	2	0	
R	2	0	
R	2	0	
W	2	3	
R	2	3	
R	29	E4	
W	29	E4	
R	29	E4	
W	28	10	
R	26	7	
W	26	7	
R	26	7	
R	37	90	
W	37	90	
R	37	90	
R	3D	2	
W	3D	2	
R	3D	2	
R	3D	2	
W	3D	2	
R	3D	2	
R	38	40	
W	38	FF	
R	38	FF	
R	3C	8F	
W	3C	8F	
R	3C	8F	
R	2A	FF	
W	2A	FF	
R	2A	FF	
R	2B	FF	
W	2B	FF	
R	2B	FF	
R	6F	30	
W	6F	30	
R	6F	30	
R	1	84	
W	1	84	
R	1	84	
W	7	D9	
W	8	0	
W	9	0	
R	1	84	
W	1	84	
R	1	84	
R	3	2	
W	3	2	
R	3	2	
R	4	9A	
W	4	9A	
R	4	9A	
R	1	84	
W	1	84	
R	1	84	
R	19	42	
W	19	42	
R	19	42	
R	1	84	
W	1	84	
R	1	84	
R	5	3	
W	5	3	
R	5	3	
R	6	33	
W	6	33	
R	6	33	
R	1	84	
W	1	84	
R	1	84	
R	11	93	
W	11	93	
R	11	93	
W	2C	0	
R	2D	2	
W	2D	2	
R	2D	2	
R	37	90	
W	37	90	
R	37	90	
R	38	FF	
W	38	40	
R	38	40	
R	2E	88	
W	2E	88	
R	2E	88	
W	2F	12	AD	
R	1	84	
W	1	84	
R	1	84	
R	2	3	
W	2	0	
R	2	0	
R	1	84	
W	1	84	
R	1	84	
R	37	90	
W	37	90	
R	37	90	
R	37	90	
W	37	90	
R	37	90	
success!
W	1	84	
R	1	84	
R	25	40	
W	25	40	
R	25	40	
R	2A	FF	
W	2A	FF	
R	2A	FF	
R	2B	FF	
W	2B	FF	
R	2B	FF	
W	27	FF	
W	28	FF	
R	13	1A	
W	13	1A	
R	13	1A	
R	5A	55	
W	5A	55	
R	5A	55	
R	5C	70	
W	5C	70	
R	5C	70	
R	1	84	
W	1	90	
R	1	90	
R	1	90	
W	1	84	
R	1	84	
R	37	90	
R	0	8	8	A	23	45	67	89	AB	CD	0	0	0	0	0	0	0	0	0	0	0	0	0	0	0	0	0	0	0	0	0	0	0	
W	27	FF	
W	28	FF	
success!
[RF69] Data:0		8 8 A 23 45 67 89 AB CD 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
```

**Expected behavior**
Receive what i sent!
> Maybe instead of having payload leght in received data, `receive` could return number of bytes (negative result would be error, positive number of bytes)

**Screenshots**

**Additional info (please complete):**
 - MCU: transmitter Adafruit Feather M0 (ATSAMD21), receiver Arduino Due
 - Link to Arduino core: [ATSAMD21](https://github.com/adafruit/ArduinoCore-samd), [DUE](https://github.com/arduino/ArduinoCore-sam)
 - Wireless module type: RFM69
 - Arduino IDE version  1.8.13
 - Library version: 4.4.0

pilotak, 2021-05-02:
This is not related to MCU, i used ESP8266 for both transmitter and receiver with the same result

jgromes, 2021-05-02:
Well this is awkward:

https://github.com/jgromes/RadioLib/blob/a32950c788a9b052dbaad34514abb581246a8aeb/src/modules/RF69/RF69.cpp#L309-L315

The packet length gets written twice, which is why you get one extra "length" byte in the packet - and also lose the last byte of actual data.

Should be fixed in the latest commit, thanks for reporting. Also, thanks for excellently prepared issue!

closed 2021-05-02 by jgromes

pilotak, 2021-05-02:
Thanks a lot, it's now working as well as AES does