jgromes/RadioLib #297: SX126x setTCXO method doesn't affect on voltage on DIO3 pin

notsleep196, 2021-05-04:
**Description**
While configuring EBYTE E22-400M30S board with SX1268 chip to work in FSK mode saw that using setTCXO() method doesn't affect on DIO3 pin of SX1268 chip. 

**To Reproduce**
Just try to configure TCXO voltage, for example:
```c++
if (state == ERR_NONE) {
  state = radio.setTCXO(1.8);
}
```

**Expected behavior**
We expect to see 1.8V supply voltage at DIO3 pin of SX1268, but it is always 1.6V (default value for this chip). Using other values leads to the same result 

**Additional info (please complete):**
 - MCU: ESP8266
 - Wireless module type SX1268
 - Library version 4.4.0

jgromes, 2021-05-04:
How did you measure the voltage? AFAIK DIO3 isn't accessible from the outside on E22-400M30S.

I guess you could also try to set some delay with `SX126x::setTCXO(voltage, delay);`, the default values is 5000 us.

notsleep196, 2021-05-04:
Thank you for quick response!
We disassembled the board by removing the protective cover to see SX1268 pins. So we used multimeter to measure voltage on DIO3.
Another approach we used is to turn on debug mode and see SPI commands. So we could see that the voltage byte in comand below is always the same whatever voltage we set in setTCXO() method. 
```
CMD	97	
DATW	0	A2	0	A2	1	A2	40	A2
```

jgromes, 2021-05-04:
Looks like I forgot to replace abs() with fabs() when detecting the voltage to be set, should be fixed in the latest commit.

jgromes, 2021-05-04:
CI builds passed for everything, looks like this is resolved.

closed 2021-05-04 by jgromes