jgromes/RadioLib #298: [SX126x] Cannot receive after transmit in FSK mode (PayloadLength is not set)

eg321, 2021-05-04:
Hi, Jan, 
It's seems we've found issue with PayloadLength in FSK mode at SX1268. Please see details below.

**Describe the bug**
It's impossible to receive  packets after any transmit.

**Actual result:**
CRC error for all received packets larger than sent one.

**My findings:**
There is "SetPacketParams" command on sx126x.  One of parameter of that command is "GFSK PacketParam7 - PayloadLength". It's meaning different for RX and TX modes:
> Size of the payload (in bytes) to transmit or maximum size of the payload that the receiver can accept.

Unfortunately RadioLib does not set it after TX (in `startReceive` method) currently.
It means that all packets larger than sent will be received with CRC error.

**Additional info (please complete):**
 - MCU: ESP8266
 - Library version 4.4.0

eg321, 2021-05-04:
As workaround:
call  `variablePacketLengthMode(255);` additionally after `startReceive()` method.

jgromes, 2021-05-04:
Thanks, I just pushed a commit that should address the issue, could you try it out in your setup?

eg321, 2021-05-05:
> Thanks, I just pushed a commit that should address the issue, could you try it out in your setup?

Commit don't looks valid:
```
RadioLib\src\modules\SX126x\SX126x.cpp:557:13: error: 'int16_t state' previously declared here
     int16_t state = setDioIrqParams(SX126X_IRQ_RX_DONE | SX126X_IRQ_TIMEOUT | SX126X_IRQ_CRC_ERR | SX126X_IRQ_HEADER_ER
R, SX126X_IRQ_RX_DONE);
```
Also I propose to avoid double call of "setPacketParams" for LoRa mode.

In general I've removed double initialization of `state` and these changes works ok now for FSK.

jgromes, 2021-05-05:
I noticed the redeclaration issue about 5 minutes after pushing the original commit, thanks to CI ;)
Also removed the redundant call to setPacketParams, thanks for pointing that out.

Should be fixed now, thanks for reporting!

closed 2021-05-05 by jgromes

eg321, 2021-05-05:
Thanks! Looking forward for release :)
Can you share ETA please?

jgromes, 2021-05-05:
Probably later today or tomorrow, there's still one open issue which I'd like to close before the next release.