jgromes/RadioLib #305: SX1278 Module bug

rserranosmith, 2021-05-18:
Hello,

I am experiencing an issue with the SX127x examples. When I declare the reset pin of my SX1278 module as pin 0, I receive the following error:

```
call of overloaded 'Module(int, int, int, int)' is ambiguous
```

The current definition of the module is as follows:

```
SX1278 loRa = new Module(1, 2, 0, 3);
```
I haven't noticed this issue with any other of my pin declarations, for example, prior to this, I had set my DIO0 pin to pin 0, and it compiled just fine. Any suggestions would be appreciated!

**Additional info:**
 - MCU: Artemis RedBoard Nano
 - Link to Arduino core: https://github.com/sparkfun/Arduino_Apollo3
 - Wireless module type: SX1278
 - Arduino IDE version: 1.8.14
 - Library version: 4.4.1

rserranosmith, 2021-05-19:
Attached is the full error. I believe that because I am defining the third entry into the module as zero, it might be confused as to whether it is a `RADIOLIB_PIN_TYPE`, or a `HardwareSerial*`.

```
In file included from /Users/ricardoserrano-smith/Documents/Arduino/libraries/RadioLib/src/RadioLib.h:44,
                 from /Users/ricardoserrano-smith/Documents/Arduino/SX127x_Receive/SX127x_Receive.ino:24:
/Users/ricardoserrano-smith/Documents/Arduino/libraries/RadioLib/src/Module.h:59:5: note: candidate: 'Module::Module(pin_size_t, pin_size_t, pin_size_t, pin_size_t)'
     Module(RADIOLIB_PIN_TYPE cs, RADIOLIB_PIN_TYPE irq, RADIOLIB_PIN_TYPE rst, RADIOLIB_PIN_TYPE gpio);
     ^~~~~~
In file included from /Users/ricardoserrano-smith/Documents/Arduino/libraries/RadioLib/src/RadioLib.h:44,
                 from /Users/ricardoserrano-smith/Documents/Arduino/SX127x_Receive/SX127x_Receive.ino:24:
/Users/ricardoserrano-smith/Documents/Arduino/libraries/RadioLib/src/Module.h:32:5: note: candidate: 'Module::Module(pin_size_t, pin_size_t, arduino::HardwareSerial*, pin_size_t)'
     Module(RADIOLIB_PIN_TYPE rx, RADIOLIB_PIN_TYPE tx, HardwareSerial* serial = &RADIOLIB_HARDWARE_SERIAL_PORT, RADIOLIB_PIN_TYPE rst = RADIOLIB_NC);
     ^~~~~~
```

jgromes, 2021-05-19:
You're correct - when the third argument is set to 0, it can be interpreted as `NULL`, which is a valid value for a `HardwareSerial*` to have (since it is a pointer). At that point, the two `Module` constructors cannot be distinguished. This is not specific to Artemis RedBoard, it happens on all platforms. 

Unfortunately I don't see an easy way out of this. In fact the only solution I can think of is to have a separate class for UART-based modules, which is going to break a lot of compatibility.

I will mark this as a bug that we're aware of, but won't fix right now (since there's a good workaround). When there are more backwards-incompatible changes, it will be fixed.

antoniosap, 2021-09-02:
you can place extra dummy argument on definition

jgromes, 2021-09-03:
@antoniosap could you elaborate? The only workaround we're currently aware of is what @rserranosmith proposed, because if you pass the 0 value by a variable, the argument won't be interpreted as NULL pointer.

antoniosap, 2021-09-03:
for example you can add a new typed dummy parameter, the compiler coupling procedure for templates will be able to distinguish the function to be called based on the type. (perfecting you can also define a typed zero) 

```
Module(RADIOLIB_PIN_TYPE cs, RADIOLIB_PIN_TYPE irq, RADIOLIB_PIN_TYPE rst, TYPE_1 dummy);
Module(RADIOLIB_PIN_TYPE cs, RADIOLIB_PIN_TYPE irq, RADIOLIB_PIN_TYPE rst, TYPE_2 dummy);
Module(RADIOLIB_PIN_TYPE cs, RADIOLIB_PIN_TYPE irq, RADIOLIB_PIN_TYPE rst, TYPE_3 dummy);
Module(RADIOLIB_PIN_TYPE cs, RADIOLIB_PIN_TYPE irq, RADIOLIB_PIN_TYPE rst, RADIOLIB_PIN_TYPE gpio);
Module(RADIOLIB_PIN_TYPE cs, RADIOLIB_PIN_TYPE irq, RADIOLIB_PIN_TYPE rst, SPIClass& spi, SPISettings spiSettings);
Module(RADIOLIB_PIN_TYPE cs, RADIOLIB_PIN_TYPE irq, RADIOLIB_PIN_TYPE rst, RADIOLIB_PIN_TYPE gpio, SPIClass& spi, SPISettings spiSettings = SPISettings(2000000, MSBFIRST, SPI_MODE0));

```

jgromes, 2021-09-04:
@antoniosap The three constructors containing dummy arguments you have proposed will still be interpreted incorrectly if `rst` is 0 - it can still be evaluated as:

`Module::Module(int, int, HardwareSerial*, int)`

With the third argument being 0, that can be treated as NULL pointer. It might resolve the ambiguity error, but instead an incorrect (or rather: unexpected) constructor will be called, UART one instead of SPI one. Instead of creating further confusion, I think it's going to be better to just break the backwards compatibility a bit, and use something like a named constructor to avoid having to create entirely new `SerialModule` class.

antoniosap, 2021-09-04:
I understand
https://stackoverflow.com/questions/7010827/named-constructor-idiom-and-new-operator

jgromes, 2021-09-05:
@antoniosap Ah, I see what you mean. This is a solution that works without breaking backwards compatibility:

```c++
enum SerialDummyEnum { serial };
enum SPIDummyEnum { spi };

(...)

Module(RADIOLIB_PIN_TYPE rx, RADIOLIB_PIN_TYPE tx, HardwareSerial* serial = nullptr, RADIOLIB_PIN_TYPE rst = RADIOLIB_NC, SerialDummyEnum dummy = serial);
Module(RADIOLIB_PIN_TYPE cs, RADIOLIB_PIN_TYPE irq, RADIOLIB_PIN_TYPE rst, RADIOLIB_PIN_TYPE gpio, SPIDummyEnum dummy = spi);
```

Neat trick, thanks!

However, I don't really like the idea of adding an extra argument just to specify which constructor to use. Seeing as the UART-based modules are not very wide-spread, I'm going to add a `SerialModule` class that inherits from `Module`.

jgromes, 2021-09-05:
Fixed in latest commit. Technically this doesn't even fully break backwards compatibility for UART-based modules (since only a convenience wrapper class was added), so I'm not going to create a major release, just a minor update or even a patch.

closed 2021-09-05 by jgromes