jgromes/RadioLib #310: LLCC68 support

xixixihaha, 2021-06-12:
the new LLCC68 based Lora
Module is out, is it possible to add the support in the coming release?

thx

**IMPORTANT: Check the wiki**
Before submitting new issue, please check the [Wiki](https://github.com/jgromes/RadioLib/wiki) and the [API documentation](https://jgromes.github.io/RadioLib/). You might find a solution to your issue there. 

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

jgromes, 2021-06-12:
Briefly going through the datasheet, I didn't see any significant difference between SX126x and LLCC68, so the current SX126x classes should work. Could you try it, as I don't have the LLCC68 module?

xixixihaha, 2021-06-13:
Thanks so much. I've verified it with SX1268 code. It worked like a charm.

jgromes, 2021-06-13:
Thanks for letting me know, I added LLCC68 as alias for SX1268, so you can use LLCC68 as any other module class:

```c++
LLCC68 radio = new Module(1, 2, 3);
```

closed 2021-06-13 by jgromes