jgromes/RadioLib #313: Testing a RFM22

lw2dtz, 2021-06-18:
Hello,

I am testing a RFM22 module and an 8mhz arduino with the library in the following modes, in RTTY and Hellschreiber it has worked as indicated. In morse it transmits but instead of having a silence between brand and brand it remains like a carrier. 
Any idea what can it be? 

Regards.

Gustavo.

jgromes, 2021-06-18:
Seems the standby method got broken at some point - not sure why. Can you try with the fixed version?

lw2dtz, 2021-06-18:
Yes, I using Radiolib 4.41 version and IDE 1.8.9

lw2dtz, 2021-06-19:
Excellent, MORSE  is solved. 
Tried AX25 mode and there is not desviation, it gives error -102 and only a tone is heard.

jgromes, 2021-06-19:
I was not able to replicate this, please follow the issue template, post the sketch you're using and the debug mode output.

lw2dtz, 2021-06-19:
I leave you the sketch and the debug file.

[ax25_rfm22.txt](https://github.com/jgromes/RadioLib/files/6681121/ax25_rfm22.txt)
[debug_rfm22.txt](https://github.com/jgromes/RadioLib/files/6681123/debug_rfm22.txt)

jgromes, 2021-06-19:
Well then it's quite obvious:

```c++
int state = radio.begin(434.0, 1.2, 0.5);
```

You've set frequency deviation to 0.5 kHz - RFM22 (or any module from the Si443x/RFM2x family) can only go down to 0.625 kHz. Remove the 0.5 and it should work.

lw2dtz, 2021-06-19:
Yes, I change it, and it doesn't work. I only get this ... no FSK ,only a carrier

[image]

jgromes, 2021-06-20:
Seems like at some point, FSK modulation config was broken, fixed now and AX.25 checked with soundmodem.

lw2dtz, 2021-06-20:
Well, the AX25 is fixed and work!!

[image]

But now the **RTTY and Hellschreiber** work bad,  it sounds as desynchronized. I'm sending you an RTTY recording.

https://user-images.githubusercontent.com/6095237/122678368-f4707680-d1bc-11eb-84ed-25f1cff2222c.mp4

jgromes, 2021-06-20:
Unforunately I was unable to reproduce this, on my end RTTY decodes fine in SDR Console and fldigi, and Hellschreiber decodes fine in fldigi.

Checking the file you sent, it looks like it's sending some really strange data. Please post the sketch you're using. Also, does this only happen on 8 MHz Arduino (I'm assuming Pro Mini)?

lw2dtz, 2021-06-20:
I posted the debug file and the skecth. I ony testing with a Pro Mini at 8 MHz.

[rtty_rfm22_ino.txt](https://github.com/jgromes/RadioLib/files/6682897/rtty_rfm22_ino.txt)

[debug_rfm22_rtty.txt](https://github.com/jgromes/RadioLib/files/6682883/debug_rfm22_rtty.txt)

jgromes, 2021-06-20:
Are you running them both with debug still enabled? If so, try disabling it. Debug mode is very slow, and in case of direct transmit modes like RTTY or Hellschreiber, it will have large impact on the timing.

lw2dtz, 2021-06-20:
You are right. Debugging mode slowed down the microcontroller. I took it out and the RTTY AND HELL started working fine.
Can SSTV mode work?  Tried but the soft goes into an infinite loop trying to initialize the RFM22.

[SSTV_RFM22.txt](https://github.com/jgromes/RadioLib/files/6683078/SSTV_RFM22.txt)

jgromes, 2021-06-21:
You might be able to do SSTV using Si443x/RFM2x, but probably only in AFSK mode (since Si443x FSK resolution is 156 Hz) and you will need something with more memory than Arduino Pro Mini.

closed 2021-06-21 by jgromes