jgromes/RadioLib #315: LLCC68 SF can't be larger than 9

xixixihaha, 2021-06-20:
According to the spec, LLCC68 support SF5~9 @ 125K BW, SF5~10 @250K BW, SF5~11 @ 500K BW
When I was playing, I found it always failed with code -20 if SF >9 when it was doing initialization.
Looks like there are still differences between SX1268 and LLCC68.

Could you help to check?
Thanks!

**IMPORTANT: Check the wiki**
Before submitting new issue, please check the [Wiki](https://github.com/jgromes/RadioLib/wiki) and the [API documentation](https://jgromes.github.io/RadioLib/). You might find a solution to your issue there. 

**Describe the bug**
A clear and concise description of what the bug is. When applicable, please include [debug mode output](https://github.com/jgromes/RadioLib/wiki/Debug-mode).

**To Reproduce**
Minimal Arduino sketch to reproduce the behavior. Please user Markdown to style the code to make it readable (see [Markdown Cheatsheet](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code)).

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional info (please complete):**
 - MCU: [e.g. Arduino Uno, ESP8266 etc.]
 - Link to Arduino core: [e.g. https://github.com/stm32duino/Arduino_Core_STM32 when using official STM32 core. See readme for links to all supported cores]
 - Wireless module type [e.g. CC1101, SX1268, etc.]
 - Arduino IDE version  [e.g. 1.8.5]
 - Library version [e.g. 3.0.0]

jgromes, 2021-06-20:
Indeed, looks like it won't be enough to just keep it as alias. I added a full class implementation for LLCC68, so now it should check bandwidth and spreading factor properly.

closed 2021-06-20 by jgromes

jgromes, 2024-05-31:
@mrx23dot I would suggest seaching the datasheet again: it does say which combinations are valid:

[image]