jgromes/RadioLib #316: Ebyte E22-400M30S module.

EliasA97, 2021-06-21:
Hello, I have this module: https://www.ebyte.com/en/product-view-news.html?id=454

In your examples it says that we can set the output power up to 22db. This module supports up to 30db or 1W output power. 
Does this library support this high power output? If not, are you able to do that?

Thank's in advance!

jgromes, 2021-06-21:
Did you read the [Wiki page](https://github.com/jgromes/RadioLib/wiki/High-power-Radio-Modules-Guide) about it? I think it explains it quite well.

EliasA97, 2021-06-21:
Well, my mistake. It seems that there is a solution there.
 Anyway, thank's for helping me!

closed 2021-06-21 by EliasA97

hjchjcjh, 2022-04-05:
I still don't understand how to turn on PA. Could you give me a detailed tutorial about turning on PA?  
Thank you very much

jgromes, 2022-04-05:
@hjchjcjh 

1. Determine which type of PA is your module using. Assuming you have E22-400M30S, that would be H0606E (probably - you should check this with the manufacturer).

2. Find the datasheet of that PA and in it, find out the gain for the frequency you are planning to use. For example, on H0606E, it will be something around 12 dB.

3. From your target power output, subtract the PA gain to figure out the output power to provide to the library. For example, for 30 dBm of total output, you will have to set power output in the library to 30 - 12 = 18 dBm.

IZ7BOJ, 2026-01-14:
Hi all,
even though a lot of time has passed, I share some test result of my colleague here.
He used a test bed composed by two lora fixed stations spaced by some Kilometers, with a stable signal.
The tx station has a E22-400M30S module.
For tx power settings greater than 18, we don't have an appreciable and proportional increse of RSSI on RX side.
Nevertheless, a power meter connected at module outputs shows these values:
- TX set at 16 --> 0,71W
- TX set at 18 --> 1,1W
- TX set at 20 --> 1,33W
- TX set at 22 --> 1,52W
This means that the output power "in excess" goes into harmonics, spurious and distorsion, and it's counterproductive.
Datasheets Absolute maximum rating is 30dBm. If the PA has 12dB gain, it means that driving signal shall be 18dBm max.
And this is confirmed by our tests.
I hope this helps,
73's
IZ7BOJ