jgromes/RadioLib #317: [ESP32] DIO actions are cleared for each "begin" call

eg321, 2021-06-21:
Hi, @jgromes 

Not sure whether that's a problem, but recently faced with surprising behavior: DIO action was cleared during changes between FSK / LoRa modes.
Additional investigation has showed that rootcause in the setting pin mode for each "begin" call (`Module::pinMode(_mod->getIrq(), INPUT);`). It looks like on ESP32 specifically "pinMode" clears interrupt on it  also.
Can't find any info to approve that behavior for ESP32. Strange that the issue is not reproduced on ESP8266.

Workaround is quite simple: we can split initialization to separate calls (`begin` / bunch of `otherInitialization` calls) or duplicate `setDio1Action` calls.

Maybe radiolib should process that difference in interrupts handling? or at least it can be documented somewhere. 

Thanks!

jgromes, 2021-06-21:
RadioLib assumes the module isn't configured prior to call to any `begin` method. The modules that have a hardware reset pin even do a full hardware reset. That's why there are calls to `pinMode` in the `begin` method.

> Can't find any info to approve that behavior for ESP32. Strange that the issue is not reproduced on ESP8266.

This behavior is very platform-dependent. Like you said, the issue isn't reproduced on ESP8266. I wouldn't be surprised if it was only present on ESP32.

I don't think this should be managed by the library. The fact that calling `pinMode()` after `attachInterrupt()` clears the interrupt service routine is completely up to the platform. What do you suggest would a fix on RadioLib side should look like?

eg321, 2021-06-23:
I think you're right. Overengineered RadioLib is not that I would to see. Let's keep interrupts handling outside of RadioLib.

closed 2021-06-23 by eg321