jgromes/RadioLib #319: RFM22 in AFSK morse

lw2dtz, 2021-06-22:
Hi,

I am testing your MORSE example in AFSK. Although I configure the audio output in the controller (Atmega328p 
@8MHZ ) and see it on the oscilloscope it is not clear where to connect on the RFM22. I guess, it must be on some
 general purpose pin ?

**IMPORTANT: Before submitting an issue, please check the following:**
1. **Read [CONTRIBUTING.md](https://github.com/jgromes/RadioLib/blob/master/CONTRIBUTING.md)!** Issues that do not follow this document will be closed/locked/deleted/ignored.
2. RadioLib has a [Wiki](https://github.com/jgromes/RadioLib/wiki) and an extensive [API documentation](https://jgromes.github.io/RadioLib/). You might find a solution to your issue there.
3. Make sure you're using the latest release of the library! Releases can be found [here](https://github.com/jgromes/RadioLib/releases).
4. Use [Arduino forums](https://forum.arduino.cc/) to ask generic questions about wireless modules, wiring, usage, etc. Only create issues for problems specific to RadioLib!
5. Error codes, their meaning and how to fix them can be found on [this page](https://jgromes.github.io/RadioLib/group__status__codes.html).

jgromes, 2021-06-22:
AFSK isn't fully supported on Si443x/RFM22 yet, because I wasn't able to find any module that had the GPIO pins connected to some pin header, so I couldn't test. In theory, you should connect the pin to GPIO2.

lw2dtz, 2021-06-22:
In practice it is the GPIO_0. It would be nice to add a txdelay and a txtail. I got this ..

https://user-images.githubusercontent.com/6095237/122977186-7900f780-d36b-11eb-8401-05514641d8c7.mp4

jgromes, 2021-06-22:
It really should be configured for GPIO2. I just tested it and found an issue with incorrect modulation being set, it should be fixed now.

closed 2021-06-22 by jgromes