jgromes/RadioLib #320: Documentation

mrquincle, 2021-06-25:
At https://github.com/jgromes/RadioLib/blob/master/src/modules/SX127x/SX1276.h#L57 it states 

```
 \param freq Carrier frequency in MHz. Allowed values range from 137.0 MHz to 525.0 MHz.
```

This thing should run on 868, so if I'm correct there's no reason to assume that for FSK this is limited? I can do a PR from https://github.com/mrquincle/RadioLib/blob/master/src/modules/SX127x/SX1276.h, but more things might need to be checked.

jgromes, 2021-06-25:
You're right, this is indeed stated incorrectly in the comment. However, the actual checks that verify this are set to the correct range (137 - 1050 MHz), so it's just a documentation issue. Feel free to submit a PR.

jgromes, 2021-06-25:
Resolved in #321

closed 2021-06-25 by jgromes