jgromes/RadioLib #326: RFM22 AFSK not cut Tx

lw2dtz, 2021-07-06:
Hi,

After its last correction in the RFM22 or Si4432 module, the problem that arose now ,
in any example is that it does not cut the carrier, once the transmission is finished.

**IMPORTANT: Check the wiki**
Before submitting new issue, please check the [Wiki](https://github.com/jgromes/RadioLib/wiki) and the [API documentation](https://jgromes.github.io/RadioLib/). You might find a solution to your issue there. 

**Describe the bug**
A clear and concise description of what the bug is. When applicable, please include [debug mode output](https://github.com/jgromes/RadioLib/wiki/Debug-mode).

**To Reproduce**
Minimal Arduino sketch to reproduce the behavior. Please user Markdown to style the code to make it readable (see [Markdown Cheatsheet](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code)).

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional info (please complete):**
 - MCU: [e.g. Arduino Uno, ESP8266 etc.]
 - Link to Arduino core: [e.g. https://github.com/stm32duino/Arduino_Core_STM32 when using official STM32 core. See readme for links to all supported cores]
 - Wireless module type [e.g. CC1101, SX1268, etc.]
 - Arduino IDE version  [e.g. 1.8.5]
 - Library version [e.g. 3.0.0]

jgromes, 2021-07-21:
I was unable to reproduce this. For example, RTTY AFSK example always correctly cuts AFSK. Please post the required information for bug report, mainly the minimal sketch that's causing the issue.

jgromes, 2021-08-03:
Closing for now - feel free to reopen later.

closed 2021-08-03 by jgromes