jgromes/RadioLib #328: random(int32_t min, int32_t max) unexpected results

tshcherban, 2021-07-12:
Is there a bug in random generate function (overload with min/max parameters)?

**To Reproduce**
```C++
SX1262 radio = new Module(NSS, DIO1, NRST, BUSY);
.
.
.
PhysicalLayer *pl = &radio;
for (int i = 0; i < 20; ++i)
{
  int32_t pause = pl->random(2500, 6500);
  Serial.println(String(pause));
}
```
**Actual result**
```
-783
2900
5732
-545
2526
4675
5544
4978
2538
5929
2069
4129
3148
4257
-3
2002
6389
6400
-942
830
```

**Expected behavior**
Should return something between specified parameters.

**Additional info (please complete):**
 - MCU: Atmega328, ESP32
 - Link to Arduino core: [e.g. https://github.com/stm32duino/Arduino_Core_STM32 when using official STM32 core. See readme for links to all supported cores]
 - Wireless module type [e.g. CC1101, SX1268, etc.]
 - Arduino IDE version 1.8.13
 - Library version 4.4.2

tshcherban, 2021-07-12:
I'm not strong in math but that line looks strange:
[image]
It does clipping the number from top, but not from bottom. Maybe use `map` or its equivalent formula here?

jgromes, 2021-07-18:
The min/max implementation is the same as in Arduino AVR core to achieve the exact same behavior as Arduino `random()` function. There was however a bug in negative number handling.

Fixed in last commit, thanks for reporting!

closed 2021-07-18 by jgromes