jgromes/RadioLib #331: LLCC68 Doesn't work

RJSDevel, 2021-07-20:
I have connected LLCC68(Ra-01sc) module to Arduino UNO, but I have got different errors from SPI
1. Initializing ... failed, code -707 - if I set BUSY pin
2. [SX1262] Scanning channel for LoRa transmission ... failed, code -20 - if I removeBUSY in constuctor
I modify emaple from master for SX1262 to:
LLCC68 radio = new Module(10, 2, 3, 9);

The first 2 module wans't connected thought level-shift, and I thought that I had broken them, but I bought new modules and level-shift YF08E, but I had got the same errors.

I don't know what to do, I did all what I knew

The changes for example RadioLib SX126x Channel Activity Detection Example:
// SX1262 has the following connections:
// NSS pin:   10
// DIO1 pin:  2
// NRST pin:  3
// BUSY pin:  9
LLCC68 radio = new Module(10, 2, 3, 9);

[image]

**Additional info (please complete):**
 - MCU: Arduino Uno
 - LLCC68
 - Arduino IDE version  [e.g. 1.8.5]
 - Library version [e.g. 3.0.0]

RJSDevel, 2021-07-20:
Add log to begin()

  int state = radio.begin();
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

[SX1262] Initializing ... reset
16:37:21.606 -> 0
16:37:21.606 -> standby
16:37:21.606 -> 0
16:37:21.606 -> config
16:37:21.606 -> 0
16:37:21.606 -> setTCXO
16:37:21.653 -> 0
16:37:21.653 -> setSpreadingFactor
16:37:21.653 -> 0
16:37:21.653 -> setBandwidth
16:37:21.700 -> 0
16:37:21.700 -> setCodingRate
16:37:21.700 -> 0
16:37:21.747 -> setSyncWord
16:37:21.747 -> 0
16:37:21.747 -> setPreambleLength
16:37:21.747 -> 0
16:37:21.747 -> setCurrentLimit
16:37:21.747 -> 0
16:37:21.747 -> setDio2AsRfSwitch
16:37:21.795 -> 0
16:37:21.795 -> setRegulatorDCDC
16:37:21.842 -> 0
16:37:21.842 -> failed, code -707

jgromes, 2021-07-20:
Depending on the position on the logging prints, it could be that the error is returned from setRegulatorDCDC. If the module is not using a DC/DC converter, but an LDO, that would explain the error. Try to call begin() with useRegulatorLDO set to false (default is true).

RJSDevel, 2021-07-20:
16:37:21.795 -> setRegulatorDCDC
16:37:21.842 -> 0 <-- Result of setRegulatorDCDC

RJSDevel, 2021-07-20:
setRegulatorLDO

[SX1262] Initializing ... reset
20:37:21.889 -> 0
20:37:21.889 -> standby
20:37:21.889 -> 0
20:37:21.889 -> config
20:37:21.889 -> 0
20:37:21.889 -> setTCXO
20:37:21.936 -> 0
20:37:21.936 -> setSpreadingFactor
20:37:21.936 -> 0
20:37:21.936 -> setBandwidth
20:37:21.983 -> 0
20:37:21.983 -> setCodingRate
20:37:21.983 -> 0
20:37:21.983 -> setSyncWord
20:37:21.983 -> 0
20:37:21.983 -> setPreambleLength
20:37:22.030 -> 0
20:37:22.030 -> setCurrentLimit
20:37:22.030 -> 0
20:37:22.030 -> setDio2AsRfSwitch
20:37:22.076 -> 0
20:37:22.076 -> setRegulatorLDO
20:37:22.076 -> 0
20:37:22.076 -> failed, code -707

RJSDevel, 2021-07-20:
I don't know how it possible that the last method returned 0, but begin returned -707

RJSDevel, 2021-07-20:
I will try to use logical analiser analyzer to see what happend in SPI

RJSDevel, 2021-07-20:
Could you say to me if answers of those functions are 0, that means SPI is ok?

jgromes, 2021-07-20:
> I will try to use logical analyzer analyzer to see what happend in SPI

Alternatively you could enable the built-in debug mode. 

There's more than just a call to SX126x::begin(), there's also a call to SX1268::begin(), it might have failed somewhere in that.

RJSDevel, 2021-07-20:
I had changed SX1262 to LLCC68

jgromes, 2021-07-20:
>I had changed SX1262 to LLCC68

That doesn't matter, since LLCC68 class still inherits from SX1262. So when you call `radio.begin()` you're actually calling `SX1262::begin()`.

RJSDevel, 2021-07-20:
You are right! 
[image]
20:56:22.033 -> setRegulatorDCDC
20:56:22.033 -> 0
20:56:22.033 -> state
20:56:22.033 -> 0
20:56:22.033 -> failed, code -707

RJSDevel, 2021-07-20:
It overrided in SX1262 but not in LLCC68

RJSDevel, 2021-07-20:
I commented it 
[image]

RJSDevel, 2021-07-20:
I have found what the deal!
[image]
setFrequency calibarations are wrong

RJSDevel, 2021-07-20:
[image]

RJSDevel, 2021-07-20:
But if send packets without interrupt appears timeout error?

RJSDevel, 2021-07-21:
The setFrequency wan't the main cause, because I'v changed not right condition

jgromes, 2021-07-21:
I don't understand why you felt the need to add `LLCC68::setFrequency()`, both `setFrequency()` and `begin()` are inherited from SX1262 class, since I wasn't able to find any difference between SX1262 and LLCC68 in that regard.

RJSDevel, 2021-07-22:
so do I, I have campare datasheets of those chips, I deleted what I changed, and clone repo again, and the chip doesn't work again, I don't know what happend, I will investigate this when I will have time

RJSDevel, 2021-07-22:
I found strange in datasheet:
[image]
[image]

jgromes, 2021-07-22:
That sort of inconsistency is fairly common in Semtech documentation, but it's not the root cause here, since we know that frequency configuration works flawlessly on all SX126x devices.

From your (fairly chaotic) posts, I can't determine whether there's an actual problem in the library at the moment, so I'm closing this issue. Feel free to open a new one once you have a reliable way to reproduce any unexpected behavior using unedited library. Also, please follow the issue template.

closed 2021-07-22 by jgromes

RJSDevel, 2021-07-26:
The same https://github.com/jgromes/RadioLib/issues/89

I will make merge request, we need to modify begin, remove tcxoVoltage from it, or override begin
tcxoVoltage have to be 0

RJSDevel, 2021-07-26:
if tcxoVoltage != 0, then calibration is failed, I connected analyser to spi and saw answer of LLCC68, it was 0xAA, this means that was calibration error

RJSDevel, 2021-08-01:
https://github.com/jgromes/RadioLib/pull/341