jgromes/RadioLib #332: [CC1101] Transmitting packet ... failed, code -5

0x5c4r3, 2021-07-20:
I am using the library and the exact example script taken from the repository (CC1101_Transmit.ino) with an ESP32 and two CC1101 - even if I have 2 wired antennas, I'm just using one for now.
I changed the pins since I am using the following:
```
CC1101 radio = new Module(27, 4, RADIOLIB_NC, 26);
```
The antenna gets configured properly, but when I try to send the message, I get this error:
```
[CC1101] Transmitting packet ... failed, code -5
```
which refers to ERR_TX_TIMEOUT  - "Timed out waiting for transmission finish."
I did not find any explaination on how to solve this issue.

Anyone have any idea?

Thanks

jgromes, 2021-07-20:
That error will occur if there's no change on the GDO0 pin within some timeout duration (5ms + 500% of the expected time-on-air). Is the connection to GDO0 secure?

0x5c4r3, 2021-07-20:
The issue seems related to the usage of 2 CC1101 on an ESP32, if I only use one -with the same pins described above - it works properly. I'll debug a bit the situation and keep you updated on what happens.

0x5c4r3, 2021-07-20:
Ok I got to the origin of the problem, I will open another issue since it's completed unrelated to this.

closed 2021-07-20 by 0x5c4r3