jgromes/RadioLib #333: ESP32 with two CC1101

0x5c4r3, 2021-07-20:
So here's the problem, I am using an esp32 with two CC1101 connected to it, using the `CC1101_Settings.ino` example. Here is my configuration:

```
CC1101 radio1 = new Module(27, 4, RADIOLIB_NC, 26);
CC1101 radio2 = new Module(13, 12, RADIOLIB_NC, 14);
```
They perfectly work alone, each one with its pin config, for instance:

```
//CC1101 radio1 = new Module(27, 4, RADIOLIB_NC, 26);
CC1101 radio2 = new Module(13, 12, RADIOLIB_NC, 14);
```
and

 ```
CC1101 radio1 = new Module(27, 4, RADIOLIB_NC, 26);
//CC1101 radio2 = new Module(13, 12, RADIOLIB_NC, 14);
```
both work perfectly, but they do not work if both used, outputting `failed, code -16` the first time, and then - if I reset the esp32 through its button - `failed, code -2`.

Debugging a bit I found out that if I disconnect the miso, mosi and sck pins of one of the two CC1101, then the script works perfectly even if I declared both of the antennas (like in the first portion of code of this post).

So so far we know that:
- the problem is not related to the pins I am using
- the problem is not related to the initial `CC1101 radio = new Module(...)` declarations.

Any ideas?

jgromes, 2021-07-20:
You're using the default constructors, which will attempt to initialize the SPI. Essentially, you're calling SPI.begin() twice and I have no idea what will that do with the SPI bus on ESP32 platform.

Try something like this:

```c++
CC1101 radio1 = new Module(27, 4, RADIOLIB_NC, 26, &SPI);
CC1101 radio2 = new Module(13, 12, RADIOLIB_NC, 14, &SPI);

void begin() {
  SPI.begin();
  radio1.begin();
  radio2.begin();
}
```

If you explicitly state the SPI instance you want to use, RadioLib won't try to initialize it automatically. Of course, you will then have to initialize the SPI yourself prior to any calls to radio1/2.

0x5c4r3, 2021-07-20:
I see, that makes sense, thanks for the quick reply!

I updated the script and now what I am getting the following error:
```
invalid conversion from 'SPIClass*' to 'uint8_t {aka unsigned char}' [-fpermissive]
```
Probably the `&SPI` in the `new Module()` call needs to get converted, not sure how.

Also, if I understand correctly, every time I need to use one of the CC1101 I need to specify which using the an SPI function, right?

jgromes, 2021-07-20:
Sorry, forgot you also have to specify SPISettings in this case. You don't have to specify anything else, just call SPI.begin() before radio.begin().

```c++
CC1101 radio1 = new Module(27, 4, RADIOLIB_NC, 26, SPI, SPISettings(2000000, MSBFIRST, SPI_MODE0));
CC1101 radio2 = new Module(13, 12, RADIOLIB_NC, 14, SPI, SPISettings(2000000, MSBFIRST, SPI_MODE0));

void begin() {
  SPI.begin();
  radio1.begin();
  radio2.begin();
}

```

0x5c4r3, 2021-07-20:
Perfectly working!
Thanks for your help and great library, I've been looking for something like this for so long!
I'll make sure to mention you and your library in my research.

Thanks again!

closed 2021-07-20 by 0x5c4r3

0x5c4r3, 2021-10-06:
Hey @jgromes , hope you alright man..
I got back to the project after a while and now seems like the double CC1101 setup that used to work (explained above) does not work anymore..
Not sure why, but I keep getting a `code -16` on the first CC1101 (and a `code 0` on the second one, but I tried many things, none of which is working).

I'm attaching the code, maybe you can see something I can't..
```
#include <RadioLib.h>

CC1101 jammer = new Module(27, 4, RADIOLIB_NC, 26, SPI, SPISettings(2000000, MSBFIRST, SPI_MODE0)); //CC1101 1
CC1101 control = new Module(13, 12, RADIOLIB_NC, 14, SPI, SPISettings(2000000, MSBFIRST, SPI_MODE0)); //CC1101 1

int state1;
int state2;

void begin(){
  SPI.begin();
    state1 = jammer.begin(433.92, 48.0, 48.0, 100.0, 10, 16);
    state2 = control.begin(433.92, 48.0, 48.0, 100.0, 10, 16);
}
 
void setup() {
  Serial.begin(9600);

  // initialize CC1101 with default settings
  Serial.println(F("[CC1101] Initializing ... "));
  begin();
 
  if (state1 == ERR_NONE and state2 == ERR_NONE) { 
    Serial.println(F("SETUP SUCCESS!"));
  } else {
    Serial.println(F("failed, codes "));
    Serial.print("state1: ");Serial.println(state1);
    Serial.print("state2: ");Serial.println(state2);
    while (true);
  }
}

void loop() {
  Serial.print(F("[CC1101] Transmitting packet ... "));

  state1 = jammer.transmit("asd");

  if (state1 == ERR_NONE) {
    // the packet was successfully transmitted
    Serial.println(F("SENT!"));

  } else if (state1 == ERR_PACKET_TOO_LONG) {
    // the supplied packet was longer than 64 bytes
    Serial.println(F("too long!"));

  } else {
    // some other error occurred
    Serial.print(F("send failed, code "));
    Serial.println(state1);

  }

  // wait for a second before transmitting again
  delay(10);
}
```

jgromes, 2021-10-07:
This is quite odd, the failed write is to enable CRC - not something that would usually fail.

Unstable output in general seems to point to issues with hardware, either bad connection or insufficient power.